ros-controls/ros2_control #1602: [Error] ros2_control tag not recognized in URDF

kbs907, 2024-07-04:
**Describe the bug**
I am currently trying to spawn a robot in Gazebo.
Although I am using the ros2_control tag in the URDF, "no ros2_control tag" error occured.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS Version : Humble

**Terminal output content**
The following content is the output displayed in the terminal when I run the launch file.
```
[INFO] [launch]: All log files can be found below /home/kbs-2204/.ros/log/2024-07-04-13-15-01-230958-kbs2204-Dell-G15-5520-6216
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [6218]
[INFO] [gzserver-2]: process started with pid [6220]
[INFO] [gzclient-3]: process started with pid [6222]
[INFO] [bash-4]: process started with pid [6224]
[INFO] [robot_control.py-5]: process started with pid [6226]
[INFO] [ros2-6]: process started with pid [6229]
[INFO] [ros2-7]: process started with pid [6231]
[INFO] [ros2-8]: process started with pid [6233]
[robot_state_publisher-1] [INFO] [1720066502.439677906] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1720066502.439795991] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1720066502.439805131] [robot_state_publisher]: got segment body_link
[robot_state_publisher-1] [INFO] [1720066502.439810789] [robot_state_publisher]: got segment body_top
[robot_state_publisher-1] [INFO] [1720066502.439815853] [robot_state_publisher]: got segment front_lidar
[robot_state_publisher-1] [INFO] [1720066502.439820733] [robot_state_publisher]: got segment front_lidar_assy
[robot_state_publisher-1] [INFO] [1720066502.439826064] [robot_state_publisher]: got segment front_rs_camera_bottom_screw_frame
[robot_state_publisher-1] [INFO] [1720066502.439831683] [robot_state_publisher]: got segment front_rs_camera_color_frame
[robot_state_publisher-1] [INFO] [1720066502.439837130] [robot_state_publisher]: got segment front_rs_camera_color_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439842360] [robot_state_publisher]: got segment front_rs_camera_depth_frame
[robot_state_publisher-1] [INFO] [1720066502.439847617] [robot_state_publisher]: got segment front_rs_camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439852632] [robot_state_publisher]: got segment front_rs_camera_left_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439857692] [robot_state_publisher]: got segment front_rs_camera_left_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439862688] [robot_state_publisher]: got segment front_rs_camera_link
[robot_state_publisher-1] [INFO] [1720066502.439867890] [robot_state_publisher]: got segment front_rs_camera_right_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439872977] [robot_state_publisher]: got segment front_rs_camera_right_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439877969] [robot_state_publisher]: got segment holder1
[robot_state_publisher-1] [INFO] [1720066502.439883209] [robot_state_publisher]: got segment holder2
[robot_state_publisher-1] [INFO] [1720066502.439888139] [robot_state_publisher]: got segment holder3
[robot_state_publisher-1] [INFO] [1720066502.439892924] [robot_state_publisher]: got segment holder4
[robot_state_publisher-1] [INFO] [1720066502.439897959] [robot_state_publisher]: got segment left_rs_camera_bottom_screw_frame
[robot_state_publisher-1] [INFO] [1720066502.439902987] [robot_state_publisher]: got segment left_rs_camera_color_frame
[robot_state_publisher-1] [INFO] [1720066502.439907856] [robot_state_publisher]: got segment left_rs_camera_color_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439912841] [robot_state_publisher]: got segment left_rs_camera_depth_frame
[robot_state_publisher-1] [INFO] [1720066502.439917901] [robot_state_publisher]: got segment left_rs_camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439923078] [robot_state_publisher]: got segment left_rs_camera_left_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439927933] [robot_state_publisher]: got segment left_rs_camera_left_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439932883] [robot_state_publisher]: got segment left_rs_camera_link
[robot_state_publisher-1] [INFO] [1720066502.439938067] [robot_state_publisher]: got segment left_rs_camera_right_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439942986] [robot_state_publisher]: got segment left_rs_camera_right_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439947962] [robot_state_publisher]: got segment rear_lidar
[robot_state_publisher-1] [INFO] [1720066502.439952665] [robot_state_publisher]: got segment rear_lidar_assy
[robot_state_publisher-1] [INFO] [1720066502.439957281] [robot_state_publisher]: got segment rear_rs_camera_bottom_screw_frame
[robot_state_publisher-1] [INFO] [1720066502.439961819] [robot_state_publisher]: got segment rear_rs_camera_color_frame
[robot_state_publisher-1] [INFO] [1720066502.439966222] [robot_state_publisher]: got segment rear_rs_camera_color_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439970853] [robot_state_publisher]: got segment rear_rs_camera_depth_frame
[robot_state_publisher-1] [INFO] [1720066502.439975470] [robot_state_publisher]: got segment rear_rs_camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439980252] [robot_state_publisher]: got segment rear_rs_camera_left_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439985018] [robot_state_publisher]: got segment rear_rs_camera_left_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.439989880] [robot_state_publisher]: got segment rear_rs_camera_link
[robot_state_publisher-1] [INFO] [1720066502.439994914] [robot_state_publisher]: got segment rear_rs_camera_right_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.439999676] [robot_state_publisher]: got segment rear_rs_camera_right_ir_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.440004622] [robot_state_publisher]: got segment right_rs_camera_bottom_screw_frame
[robot_state_publisher-1] [INFO] [1720066502.440009325] [robot_state_publisher]: got segment right_rs_camera_color_frame
[robot_state_publisher-1] [INFO] [1720066502.440014319] [robot_state_publisher]: got segment right_rs_camera_color_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.440722785] [robot_state_publisher]: got segment right_rs_camera_depth_frame
[robot_state_publisher-1] [INFO] [1720066502.440754506] [robot_state_publisher]: got segment right_rs_camera_depth_optical_frame
[robot_state_publisher-1] [INFO] [1720066502.440794426] [robot_state_publisher]: got segment right_rs_camera_left_ir_frame
[robot_state_publisher-1] [INFO] [1720066502.440800132] [robot_state_publisher]: got segment right_rs_camera_left_ir_optical_frame
[... 212 lines cut ...]
            cmd=['ros2', 'control', 'load_controller', '--set-state', 'active', 'forward_velocity_controller'], output='screen'
        )

    joint_state_broadcaster = ExecuteProcess(
            cmd=['ros2', 'control', 'load_controller', '--set-state', 'active', 'joint_state_broadcaster'], output='screen'
        )

    nodes = [
        gazebo_model_path,
        robot_state_publisher_node,
        gzserver_cmd,
        gzclient_cmd,
        spawn_robot_node,
        controller,
        forward_position_controller,
        forward_velocity_controller,
        joint_state_broadcaster
    ]
    
    return LaunchDescription(nodes)
```

Even though I am using the ros2_control tag, I am experiencing this error, which is causing me a lot of distress. 
I have tried checking and attempting various things, but I am still unable to figure out the root cause. 
If anyone could help me, I would be extremely grateful.

kbs907, 2024-07-04:
Oh my god.. I removed the 'gazebo' tag that was wrapping it, and now it started recognizing the 'ros2_control' tag.

closed 2024-07-04 by kbs907

bmagyar, 2024-07-04:
Sometimes you only need a rubber duck to explain a problem to ;)