ros-controls/ros2_control #1610: Overlay build does not work with rolling/jammy

christophfroehlich, 2024-07-13:
Yes, because since a couple of months there is no rolling release to 22.04 any more -> there was a switch to noble 24.04. I'm not sure what we can do to avoid such a confusion. @saikishor a compile-time error if the wrong version of hardware_interface_testing is to be built?

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_control/issues/1609#issuecomment-2225102337_

saikishor, 2024-07-16:
I've tried using jammy and compiling it from source and it works. I have been working on Jammy for last 2 months.

christophfroehlich, 2024-07-16:
but you have installed the rolling jammy binaries?

saikishor, 2024-07-16:
> but you have installed the rolling jammy binaries?

Yes, my earlier docker was with Rolling Jammy, and the binaries it had were up-to-date at least

christophfroehlich, 2024-07-16:
@Parker-Drouillard you had the rolling binaries installed, right? Which exact version? `$ apt show ros-rolling-ros2-control ros-rolling-hardware-interface-testing `

If this remains a problem, could you provide a docker image where you can reproduce your issue?

Parker-Drouillard, 2024-07-16:
@christophfroehlich 

I'll look into this more tonight though and see if I am still having the issues & if so can set up the docker image. 

My exact output from your command is as follows:

```
Package: ros-rolling-ros2-control
Version: 4.5.0-1jammy.20240216.192328
Priority: optional
Section: misc
Maintainer: Bence Magyar <bence.magyar.robotics@gmail.com>
Installed-Size: 43.0 kB
Depends: ros-rolling-controller-interface, ros-rolling-controller-manager, ros-rolling-controller-manager-msgs, ros-rolling-hardware-interface, ros-rolling-joint-limits, ros-rolling-ros2-control-test-assets, ros-rolling-ros2controlcli, ros-rolling-transmission-interface, ros-rolling-ros-workspace
Download-Size: 6,342 B
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Metapackage for ROS2 control related packages

Package: ros-rolling-hardware-interface-testing
Version: 4.5.0-1jammy.20240216.184930
Priority: optional
Section: misc
Maintainer: Bence Magyar <bence.magyar.robotics@gmail.com>
Installed-Size: 147 kB
Depends: libc6 (>= 2.14), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libstdc++6 (>= 11), ros-rolling-control-msgs, ros-rolling-hardware-interface, ros-rolling-lifecycle-msgs, ros-rolling-pluginlib, ros-rolling-rclcpp-lifecycle, ros-rolling-ros2-control-test-assets, ros-rolling-ros-workspace
Download-Size: 32.7 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: ros2_control hardware interface testing
```

closed 2025-05-16 by github-actions[bot]