ros-controls/ros2_control #1615: Can I use Python to fully write the Ros2 control package?

zouzhe1, 2024-07-16:
Can I use Python to fully write the Ros2 control package? 

Because I am not familiar with CPP,
 and I am used to using Python libraries such as can, canopen, and so on.

And my Python program for controlling hardware has already been written

christophfroehlich, 2024-07-16:
No, currently this is not possible. There were some attempts for python bindings for controllers afair, maybe @olivier-stasse can give an update on this topic.

If you don't have hard realtime requiremtents, you can use a [topic based system](https://github.com/PickNikRobotics/topic_based_ros2_control) with existing python hardware drivers.

olivier-stasse, 2024-07-16:
Dear @zouzhe1 ,
There is no solution which fits all the problems.

The ros2_controllers are already covering a large part of real life robotics problem.

But if you are designing your own controller and it is quite complex, I would strongly advice to decouple the mathematics from the encapsulation. You can view the ros2_controller interface as a standard interface to encapsulate controllers.

If you are quite good at python you can try:
https://github.com/ICube-Robotics/pytroller
generating a ros2_controller from python.

For complex controller, such as inverse dynamics on humanoid robot, and quick prototyping it is usually recommended to check the math with python following examples such as:
https://github.com/stack-of-tasks/tsid/tree/master/exercizes if you want to do Task Space Inverse Dynamics.

Then you can deploy it in a custom ros2_controller.

We wrote some python bindings to ros2_control, but they need more work.

closed 2025-05-16 by github-actions[bot]