ros-controls/ros2_control #1616: Unable to fetch binaries for Iron distro

anath93, 2024-07-16:
**Expected behavior**
ros2 package page show that these packages are released
https://index.ros.org/r/ros2_controllers/github-ros-controls-ros2_controllers/#iron

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Version Iron

christophfroehlich, 2024-07-16:
which ubuntu are you using?

anath93, 2024-07-16:
22.04

anath93, 2024-07-16:
@christophfroehlich, would be better to build locally for now or simple fix ?

christophfroehlich, 2024-07-16:
have you set the apt-list properly and run apt-get update afterwards?

anath93, 2024-07-16:
Yes ! All the other packages are good except ros2_control and ros2_controllers.

anath93, 2024-07-17:
Its recently upstream issue on recent sync of iron for ARM64 platform.

christophfroehlich, 2024-07-17:
Thanks for reporting.
https://robotics.stackexchange.com/questions/112249/ros2-iron-arm64-packages-vanished
https://discourse.ros.org/t/ros2-iron-arm64-packages-vanished/38652/2

Any idea what happened @bmagyar ?

christophfroehlich, 2024-07-17:
However, the last build [was successful](https://build.ros2.org/view/Ibin_ujv8_uJv8/job/Ibin_ujv8_uJv8__controller_interface__ubuntu_jammy_arm64__binary/), hopefully this solves itself with the next iron sync. @anath93 you can try the [testing repo](https://docs.ros.org/en/iron/Installation/Testing.html) in the meantime.

anath93, 2024-07-17:
The sync took place and packages back to there current release. 

https://discourse.ros.org/t/new-packages-for-iron-irwini-2024-07-17/38674

closed 2024-07-17 by anath93