ros-controls/ros2_control #1617: Transmission Interface via Functional Mock-up Interface (FMI)

olneumann, 2024-07-17:
Would it be possible to add an interface via the transmission interface that allows dynamic/kinematic models generated via the FMI standard to be integrated? Particularly for more complex drive systems with tendons, serial springs, etc., modelling, e.g. in Modelica, would be expedient in order to then simply integrate the compiled dynamic model as an interface between the joint angle and actuator(s).

I could imagine adding this as an option in the transmission interface, similar to the approaches below in `macro.ros2_control.xacro`:

```xml
<transmission name="transmission1">
  <plugin>transmission_model_name</plugin>
  <fmu>model.fmu</plugin>
  <actuator name 1="actuator1" role="actuator1"/>
  <actuator name 2="actuator2" role="actuator2"/>
  <joint name="joint1" role="joint1">
    <params>some_params</params>
  </joint>
</transmission>
```

Further links on this topic:

- [gazebo-fmi actuator plugin](https://github.com/robotology/gazebo-fmi/tree/master/plugins/actuator)
- [fmi_adapter for ROS nodes](https://github.com/boschresearch/fmi_adapter)

bmagyar, 2024-07-17:
Thank you, this is exactly where we envisioned that part of the framework to go!

Do you have / know of existing work implementing it?

olneumann, 2024-07-17:
Beside the [fmi_adapter](https://github.com/boschresearch/fmi_adapter/tree/rolling) from @ralph-lange and collegues I didn't found any implementation for ROS2. But [here](https://github.com/traversaro/awesome-fmi?tab=readme-ov-file) is a list of libraries, tools and resources.

With the adapter you can spawn a node which takes scalar float inputs and outputs and maps those onto individual subscriber and publisher. For this I currently do a hack and map those with another node onto the` <controller>/commands` as the adapter those not support float arrays. But this has the drawback that I cannot easily add another trajectory controller on top of this (without remapping again...). Doing this in the xacro file would be great.

For reference like:

``` xml
<ros2_control name="fmi_integration_example" type="system">
  <hardware>
    <plugin>actuator_hardware_plugin_name</plugin>
    <param name="param">42</param>
  </hardware>

  <joint name="actuator0">
          <param name="id">0</param>
          <command_interface name="effort"/>
          <state_interface name="position"/>
          <state_interface name="velocity"/>
          <state_interface name="effort"/>
  </joint>

  <joint name="actuator1">
          <param name="id">1</param>
          <command_interface name="effort"/>
          <state_interface name="position"/>
          <state_interface name="velocity"/>
          <state_interface name="effort"/>
  </joint>

  <transmission name="transmission1">
    <plugin>some_fmi_plugin</plugin>
    <fmu>fmu_path</plugin>
    <param name="joint">{joint1}</param>
    <param name="actuator">{actuator1, actuator2}</param>
    <param name="some_passed_param">42</param>
  </transmission>
</ros2_control>
```

As a side questions, are there any plans to be able to distinguish between joints and actuators in the above definition, i.e. with a dedicated <actuator name=""</acutator> definition?

closed 2025-05-16 by github-actions[bot]