ros-controls/ros2_control #1618: Rolling compatibility on Iron seems to be broken

christophfroehlich, 2024-07-17:
https://github.com/ros-controls/ros2_control_ci/actions/runs/9966931009/job/27539804388

@saikishor any ideas what that means?

```
Error: ROR] [1721182624.941512025] [test_controller_manager.resource_manager]: Exception occurred while initializing hardware 'TestActuatorHardware': failed to call rcl_logging_rosout_add_sublogger: The entry of logger 'test_controller_manager.resource_manager' not exist., at ./src/rcl/logging_rosout.c:438
```

christophfroehlich, 2024-07-17:
and why it doesnt fail on humble or jazzy?

saikishor, 2024-07-17:
@christophfroehlich This seems to be the recently added logger stuff in the ResourceManager. It is really weird that it fails only on Iron

closed 2024-11-22 by christophfroehlich