ros-controls/ros2_control #1632: Avoid calling write in inactive state

christophfroehlich, 2024-07-20:
After looking at the code, I believe this is happening because we call the `write` method also for the `inactive` components 
https://github.com/ros-controls/ros2_control/blob/master/hardware_interface%2Fsrc%2Fsystem.cpp#L247 and https://github.com/ros-controls/ros2_control/blob/master/hardware_interface%2Fsrc%2Factuator.cpp#L251

If we need to properly fix it, I think we might need a new method called `dynamic_configure` where all the `gpio` interfaces go in and then the main hardware command interfaces can go in the `write` method, in this way we can call this new `dynamic_configure` method and avoid calling `write` method in the `inactive` state.

_Originally posted by @saikishor in https://github.com/ros-controls/ros2_control_demos/issues/455#issuecomment-1974937471_

saikishor, 2024-07-30:
The above should be fixed by https://github.com/ros-controls/ros2_control/pull/1451

christophfroehlich, 2024-08-02:
> The above should be fixed by #1451

I can confirm this by removing the `std::is_finite` check from example_14: it doesn't fail now any more, but does with reverting 25f2c97eb909c48b96384c7cff7f7f36be8ca509

closed 2024-08-02 by christophfroehlich

saikishor, 2024-08-02:
Thanks a lot for crosschecking @christophfroehlich. You are the best