ros-controls/ros2_control #1648: /controller_manager is not loading up

nisathav, 2024-07-31:
**Describe the bug**
I am trying to run a differential drive robot in my computer. The controls are working fine when used gazebo_controls but when I am switching to ros2_controls, the following errors are received. I am using ros2 humble version.

"nisat@nisat:~/Desktop/ros2_ws$ ros2 launch mangobee_differentialdrivewheel_robot launch_sim.launch.py 
[INFO] [launch]: All log files can be found below /home/nisat/.ros/log/2024-07-31-16-11-02-560368-nisat-20693
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [20697]
[INFO] [twist_mux-2]: process started with pid [20699]
[INFO] [gzserver-3]: process started with pid [20701]
[INFO] [gzclient-4]: process started with pid [20703]
[INFO] [spawn_entity.py-5]: process started with pid [20705]
[INFO] [spawner-6]: process started with pid [20707]
[INFO] [spawner-7]: process started with pid [20709]
[robot_state_publisher-1] [INFO] [1722422463.063742555] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1722422463.063798690] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1722422463.063803619] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1722422463.063806124] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-1] [INFO] [1722422463.063808478] [robot_state_publisher]: got segment caster_wheel
[robot_state_publisher-1] [INFO] [1722422463.063810718] [robot_state_publisher]: got segment chassis
[robot_state_publisher-1] [INFO] [1722422463.063813059] [robot_state_publisher]: got segment laser_frame
[robot_state_publisher-1] [INFO] [1722422463.063815196] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1722422463.063817373] [robot_state_publisher]: got segment right_wheel
[twist_mux-2] [INFO] [1722422463.401782829] [twist_mux]: Topic handler 'topics.joystick' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 100.
[twist_mux-2] [INFO] [1722422463.403815181] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 10.
[twist_mux-2] [INFO] [1722422463.404142094] [twist_mux]: Topic handler 'topics.tracker' subscribed to topic 'cmd_vel_tracker': timeout = 0.500000s , priority = 20.
[spawn_entity.py-5] [INFO] [1722422463.434962560] [spawn_entity]: Spawn Entity started
[spawn_entity.py-5] [INFO] [1722422463.435144421] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-5] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-5]   warnings.warn(
[spawn_entity.py-5] [INFO] [1722422463.436495744] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-5] [INFO] [1722422463.447771137] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-5] [INFO] [1722422463.447998624] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-5] [INFO] [1722422463.952209711] [spawn_entity]: Calling service /spawn_entity
[gzserver-3] [INFO] [1722422464.780959597] [camera_controller]: Publishing camera info to [/camera/camera_info]
[gzserver-3] [WARN] [1722422464.786500354] [rcl]: Found remap rule '~/out:=scan'. This syntax is deprecated. Use '--ros-args --remap ~/out:=scan' instead.
[gzserver-3] [WARN] [1722422464.788452744] [rcl]: Found remap rule '~/out:=scan'. This syntax is deprecated. Use '--ros-args --remap ~/out:=scan' instead.
[spawn_entity.py-5] [INFO] [1722422464.881699912] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [mangobee_differentialdrivewheel_robot]
[INFO] [spawn_entity.py-5]: process has finished cleanly [pid 20705]
[spawner-7] [INFO] [1722422465.440287175] [spawner_diff_cont]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1722422465.442477638] [spawner_joint_broad]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1722422467.452992102] [spawner_diff_cont]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1722422467.454625794] [spawner_joint_broad]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1722422469.468981534] [spawner_diff_cont]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1722422469.469326153] [spawner_joint_broad]: Waiting for '/controller_manager' node to exist
[spawner-6] [INFO] [1722422471.483041239] [spawner_joint_broad]: Waiting for '/controller_manager' node to exist
[spawner-7] [INFO] [1722422471.483713395] [spawner_diff_cont]: Waiting for '/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1722422471.595004976] [rclcpp]: signal_handler(signum=2)
[twist_mux-2] [INFO] [1722422471.595089902] [rclcpp]: signal_handler(signum=2)
[spawner-6] Traceback (most recent call last):
[spawner-6]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/lib/controller_manager/spawner", line 33, in <module>
[spawner-6]     sys.exit(load_entry_point('controller-manager==2.42.0', 'console_scripts', 'spawner')())
[spawner-6]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-6]     if not wait_for_controller_manager(
[spawner-6]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 106, in wait_for_controller_manager
[spawner-6]     node_and_namespace = wait_for_value_or(
[spawner-6]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 66, in wait_for_value_or
[spawner-6]     time.sleep(0.2)
[spawner-6] KeyboardInterrupt
[spawner-7] Traceback (most recent call last):
[spawner-7]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/lib/controller_manager/spawner", line 33, in <module>
[spawner-7]     sys.exit(load_entry_point('controller-manager==2.42.0', 'console_scripts', 'spawner')())
[spawner-7]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-7]     if not wait_for_controller_manager(
[spawner-7]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 106, in wait_for_controller_manager
[spawner-7]     node_and_namespace = wait_for_value_or(
[spawner-7]   File "/home/nisat/Desktop/ros2_ws/install/controller_manager/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 66, in wait_for_value_or
[spawner-7]     time.sleep(0.2)
[spawner-7] KeyboardInterrupt
[ERROR] [spawner-7]: process has died [pid 20709, exit code -2, cmd '/home/nisat/Desktop/ros2_ws/install/controller_manager/lib/controller_manager/spawner diff_cont --controller-manager /controller_manager --ros-args'].
[... 114 lines cut ...]
    
  Here is my controller yaml file,
  "controller_manager:
  ros__parameters:
    update_rate: 100.0
    use_sim_time: true

    diff_cont:
      type: diff_drive_controller/DiffDriveController
    
    joint_broad:
      type: joint_state_broadcaster/JointStateBroadcaster

diff_cont:
  ros__parameters:

    publish_rate: 100.0 
    base_frame_id: base_link

    left_wheel_names: ['left_wheel_joint']
    right_wheel_names: ['right_wheel_joint']
    wheel_separation: 0.35
    wheel_radius: 0.05

    use_stamped_vel: false"

christophfroehlich, 2024-07-31:
Please have a look at our [demos](https://control.ros.org/humble/doc/ros2_control_demos/example_1/doc/userdoc.html). You need to run the controller_manager from the ros2_control_node, which is done by gazebo in the gazebo_ros2_control plugin.

sees1, 2024-07-31:
ummm... He did it. He parse xacro file which include ros2_control tag along with gazebo_ros2_control plugin.
robot.urdf.xacro
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro"  name="robot">

    <xacro:arg name="use_ros2_control" default="true"/>
    <xacro:arg name="sim_mode" default="false"/>

    <xacro:include filename="robot_core.xacro" />

    <xacro:if value="$(arg use_ros2_control)">
        <xacro:include filename="ros2_control.xacro" />
    </xacro:if>
    <xacro:unless value="$(arg use_ros2_control)">
        <xacro:include filename="gazebo_control.xacro" />
    </xacro:unless>
    <xacro:include filename="lidar.xacro" />
    <xacro:include filename="camera.xacro" />
    <!-- <xacro:include filename="depth_camera.xacro" /> -->

    <xacro:include filename="face.xacro" />
    
</robot>
```
and here ros2_control.xacro
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:unless value="$(arg sim_mode)">
        <ros2_control name="RealRobot" type="system">
            <hardware>
                <plugin>diffdrive_arduino/DiffDriveArduinoHardware</plugin>
                <param name="left_wheel_name">left_wheel_joint</param>
                <param name="right_wheel_name">right_wheel_joint</param>
                <param name="loop_rate">30</param>
                <param name="device">/dev/ttyUSB0</param>
                <param name="baud_rate">57600</param>
                <param name="timeout_ms">1000</param>
                <param name="enc_counts_per_rev">3436</param>
            </hardware>
            <joint name="left_wheel_joint">
                <command_interface name="velocity">
                    <param name="min">-10</param>
                    <param name="max">10</param>
                </command_interface>
                <state_interface name="position"/>
                <state_interface name="velocity"/>
            </joint>
            <joint name="right_wheel_joint">
                <command_interface name="velocity">
                    <param name="min">-10</param>
                    <param name="max">10</param>
                </command_interface>
                <state_interface name="position"/>
                <state_interface name="velocity"/>
            </joint>
        </ros2_control>
    </xacro:unless>

    <xacro:if value="$(arg sim_mode)">
        <ros2_control name="GazeboSystem" type="system">
            <hardware>
                <plugin>gazebo_ros2_control/GazeboSystem</plugin>
            </hardware>
            <joint name="left_wheel_joint">
                <command_interface name="velocity">
                    <param name="min">-10</param>
                    <param name="max">10</param>
                </command_interface>
                <state_interface name="velocity"/>
                <state_interface name="position"/>
            </joint>
            <joint name="right_wheel_joint">
                <command_interface name="velocity">
                    <param name="min">-10</param>
                    <param name="max">10</param>
                </command_interface>
                <state_interface name="velocity"/>
                <state_interface name="position"/>
            </joint>
        </ros2_control>
    </xacro:if>

    <gazebo>
        <plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">
            <parameters>$(find articubot_one)/config/my_controllers.yaml</parameters>
            <parameters>$(find articubot_one)/config/gaz_ros2_ctl_use_sim.yaml</parameters>
        </plugin>
    </gazebo>

</robot>
```
for full code check repo: https://github.com/joshnewans/articubot_one. i think that code in question from here.

christophfroehlich, 2024-08-01:
> ummm... He did it. He parse xacro file which include ros2_control tag along with gazebo_ros2_control plugin. robot.urdf.xacro

No he did not? At least from the launch files he added here. See the launch_robot.launch.py in the linked repo

https://github.com/joshnewans/articubot_one/blob/cd1eb0f503012a65552fd088cdbd9cf8ac133cbc/launch/launch_robot.launch.py#L53-L58

sees1, 2024-08-01:
Maybe he didn't install gazebo_ros2_control package? Because, when i check install, i didn't find it, after installing my error is gone, and controller_manager load.
But, you point to a launch, that make sense on real robot, not for gazebo simulation. Here 2 launch file, launch_sim.launch.py and launch_robot.launch.py
P.S. Sorry for my bad english.

Sleepytools4321, 2024-08-02:
os2 launch articubot_one launch_robot.launch.py
[INFO] [launch]: All log files can be found below /home/aryan/.ros/log/2024-08-02-12-17-52-322344-aryan-devros-53176
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [53179]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link chassis had 4 children
[robot_state_publisher-1] Link camera_link had 1 children
[robot_state_publisher-1] Link camera_link_optical had 0 children
[robot_state_publisher-1] Link caster_wheel had 0 children
[robot_state_publisher-1] Link face_link had 0 children
[robot_state_publisher-1] Link laser_frame had 0 children
[robot_state_publisher-1] Link left_wheel had 0 children
[robot_state_publisher-1] Link right_wheel had 0 children
[robot_state_publisher-1] [INFO] [1722581272.571053880] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1722581272.571218780] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1722581272.571254706] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-1] [INFO] [1722581272.571269389] [robot_state_publisher]: got segment caster_wheel
[robot_state_publisher-1] [INFO] [1722581272.571280558] [robot_state_publisher]: got segment chassis
[robot_state_publisher-1] [INFO] [1722581272.571287627] [robot_state_publisher]: got segment face_link
[robot_state_publisher-1] [INFO] [1722581272.571294495] [robot_state_publisher]: got segment laser_frame
[robot_state_publisher-1] [INFO] [1722581272.571300960] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1722581272.571310657] [robot_state_publisher]: got segment right_wheel
[INFO] [ros2_control_node-2]: process started with pid [53244]
[INFO] [spawner.py-3]: process started with pid [53246]
[INFO] [spawner.py-4]: process started with pid [53248]
[ros2_control_node-2] [INFO] [1722581277.170889793] [DiffDriveArduino]: Configuring...
[ros2_control_node-2] [INFO] [1722581277.171403217] [DiffDriveArduino]: Finished Configuration
[ros2_control_node-2] [INFO] [1722581277.171547416] [DiffDriveArduino]: Starting Controller...
[ros2_control_node-2] [INFO] [1722581278.187907705] [controller_manager]: update rate is 30 Hz
[ros2_control_node-2] [INFO] [1722581278.385951662] [controller_manager]: Loading controller 'diff_cont'
[ros2_control_node-2] [ERROR] [1722581278.386027887] [controller_manager]: Loader for controller 'diff_cont' not found.
[ros2_control_node-2] [INFO] [1722581278.386053803] [controller_manager]: Available classes:
[ros2_control_node-2] [INFO] [1722581278.386076364] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-2] [INFO] [1722581278.386095316] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-2] [INFO] [1722581278.386113955] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ros2_control_node-2] [INFO] [1722581278.386238601] [controller_manager]: Loading controller 'joint_broad'
[ros2_control_node-2] [ERROR] [1722581278.386289813] [controller_manager]: Loader for controller 'joint_broad' not found.
[ros2_control_node-2] [INFO] [1722581278.386314716] [controller_manager]: Available classes:
[ros2_control_node-2] [INFO] [1722581278.386329189] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-2] [INFO] [1722581278.386339721] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-2] [INFO] [1722581278.386349637] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ERROR] [spawner.py-3]: process has died [pid 53246, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint_broad --ros-args'].
[ERROR] [spawner.py-4]: process has died [pid 53248, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py diff_cont --ros-args'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ros2_control_node-2] [INFO] [1722581368.768283810] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-1] [INFO] [1722581368.768321864] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 53179]
[INFO] [ros2_control_node-2]: process has finished cleanly [pid 53244]

@christophfroehlich Facing a similar issue with ROS-2 Contoroller, could you please look into it

BengIslam7, 2025-03-03:
**Solution :**
joint_state_broadcaster = TimerAction(
        period=7.0,  # Delay of 3 seconds before starting joint_state_broadcaster
        actions=[
            Node(
                package='controller_manager',
                executable='spawner',
                output='screen',
                arguments=["joint_broad"]
            )
        ]
    )

    diff_drive_controller = TimerAction(
        period=10.0,  # Delay of 5 seconds before starting forward_position_controller
        actions=[
            Node(
                package='controller_manager',
                executable='spawner',
                output='screen',
                arguments=["diff_cont"]
            )
        ]
    )

christophfroehlich, 2025-03-04:
@BengIslam7 solution for what? instead, the spawner has parameters to increase different timeouts, see [the docs](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#spawner). This might be needed if the simulation takes longer to start up.

christophfroehlich, 2025-03-04:
Closing due to inactivity of the original post.

closed 2025-03-04 by christophfroehlich