ros-controls/ros2_control #1655: os2 launch articubot_one launch_robot.launch.py

Sleepytools4321, 2024-08-02:
os2 launch articubot_one launch_robot.launch.py
[INFO] [launch]: All log files can be found below /home/aryan/.ros/log/2024-08-02-12-17-52-322344-aryan-devros-53176
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [53179]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link chassis had 4 children
[robot_state_publisher-1] Link camera_link had 1 children
[robot_state_publisher-1] Link camera_link_optical had 0 children
[robot_state_publisher-1] Link caster_wheel had 0 children
[robot_state_publisher-1] Link face_link had 0 children
[robot_state_publisher-1] Link laser_frame had 0 children
[robot_state_publisher-1] Link left_wheel had 0 children
[robot_state_publisher-1] Link right_wheel had 0 children
[robot_state_publisher-1] [INFO] [1722581272.571053880] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1722581272.571218780] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-1] [INFO] [1722581272.571254706] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-1] [INFO] [1722581272.571269389] [robot_state_publisher]: got segment caster_wheel
[robot_state_publisher-1] [INFO] [1722581272.571280558] [robot_state_publisher]: got segment chassis
[robot_state_publisher-1] [INFO] [1722581272.571287627] [robot_state_publisher]: got segment face_link
[robot_state_publisher-1] [INFO] [1722581272.571294495] [robot_state_publisher]: got segment laser_frame
[robot_state_publisher-1] [INFO] [1722581272.571300960] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-1] [INFO] [1722581272.571310657] [robot_state_publisher]: got segment right_wheel
[INFO] [ros2_control_node-2]: process started with pid [53244]
[INFO] [spawner.py-3]: process started with pid [53246]
[INFO] [spawner.py-4]: process started with pid [53248]
[ros2_control_node-2] [INFO] [1722581277.170889793] [DiffDriveArduino]: Configuring...
[ros2_control_node-2] [INFO] [1722581277.171403217] [DiffDriveArduino]: Finished Configuration
[ros2_control_node-2] [INFO] [1722581277.171547416] [DiffDriveArduino]: Starting Controller...
[ros2_control_node-2] [INFO] [1722581278.187907705] [controller_manager]: update rate is 30 Hz
[ros2_control_node-2] [INFO] [1722581278.385951662] [controller_manager]: Loading controller 'diff_cont'
[ros2_control_node-2] [ERROR] [1722581278.386027887] [controller_manager]: Loader for controller 'diff_cont' not found.
[ros2_control_node-2] [INFO] [1722581278.386053803] [controller_manager]: Available classes:
[ros2_control_node-2] [INFO] [1722581278.386076364] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-2] [INFO] [1722581278.386095316] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-2] [INFO] [1722581278.386113955] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ros2_control_node-2] [INFO] [1722581278.386238601] [controller_manager]: Loading controller 'joint_broad'
[ros2_control_node-2] [ERROR] [1722581278.386289813] [controller_manager]: Loader for controller 'joint_broad' not found.
[ros2_control_node-2] [INFO] [1722581278.386314716] [controller_manager]: Available classes:
[ros2_control_node-2] [INFO] [1722581278.386329189] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-2] [INFO] [1722581278.386339721] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-2] [INFO] [1722581278.386349637] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ERROR] [spawner.py-3]: process has died [pid 53246, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py joint_broad --ros-args'].
[ERROR] [spawner.py-4]: process has died [pid 53248, exit code 1, cmd '/opt/ros/foxy/lib/controller_manager/spawner.py diff_cont --ros-args'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ros2_control_node-2] [INFO] [1722581368.768283810] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-1] [INFO] [1722581368.768321864] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 53179]
[INFO] [ros2_control_node-2]: process has finished cleanly [pid 53244]

@christophfroehlich Facing a similar issue with ROS-2 Contoroller, could you please look into it

_Originally posted by @Sleepytools4321 in https://github.com/ros-controls/ros2_control/issues/1648#issuecomment-2264719664_

christophfroehlich, 2024-08-02:
This is different to #1648 , please read the log carefully. 

> [ros2_control_node-2] [ERROR] [1722581278.386289813] [controller_manager]: Loader for controller 'joint_broad' not found.
> [ros2_control_node-2] [INFO] [1722581278.386314716] [controller_manager]: [...]

Have you installed ros2_controllers?
Be aware that foxy distro is EOL for a long time now, and we can't give any support for that any more.

closed 2024-08-02 by christophfroehlich

Sleepytools4321, 2024-08-02:
First of all apologies for putting a comment here, thought this issue was relevant to mine. I did install ros2_controllers, I pulled the repo from Josh's repo. By any chance you would be able to help with this error. I am not sure but this seems like a std missing file error to me. Again Sorry about the old comment and thanks in advance

christophfroehlich, 2024-08-02:
Please update your distro to at least humble.

Sleepytools4321, 2024-08-02:
Okay let me try that. Thanks