ros-controls/ros2_control #1656: Tell hardware_interface which state interfaces to populate

coxep, 2024-08-03:
**Is your feature request related to a problem? Please describe.**
I have a hardware interface that exposes multiple interfaces that would traditionally live in other parts of a robotics stack. But don't want to provide these interfaces unless they are needed by a controller/broadcaster. I.e., if I load a battery state broadcaster, I would want the hardware interface to start reading battery/voltage, battery/current, etc. But otherwise, it is wasteful to read these.

**Describe the solution you'd like**
Some method (similar to prepare_command_mode_switch) would be nice to tell the hardware_interface what state interfaces to populate

**Describe alternatives you've considered**
I can add parameters to hardware interface describing which state interfaces should be populated, and manually ensure these match state interfaces needed by controllers/broadcasters.

**Additional context**
The hardware interface wraps a single connection to the underlying hardware, which is an embedded system containing multiple states, controllers, and sensors.

christophfroehlich, 2024-08-03:
What is the benefit of not reading it if nothing "subscribes"? From a through-put perspective, the component should be anyways able to read everything within one update cycle?

coxep, 2024-08-03:
Very fair point. OK, closing this.

closed 2024-08-03 by coxep