ros-controls/ros2_control #1702: Add tests to the `params_file` parameter in test_spawner_unspawner.cpp

saikishor, 2024-08-20:
There was a typo observed after using #1661 and it seems like there is no proper tests with test_controller that uses the parameters from param_file, that's the reason it is not detected in the first place. It is fixed in the PR: #1701

We need to add some tests here to avoid such cases in future

SantoshGovindaraj, 2024-09-15:
Hello @saikishor , I would like to contribute to this issue as my first step in open-source. Here is my plan for the tests:
* Add test cases for passing a valid `params_file` and verifying that the controller is loaded with the expected outcomes.
* Test with an invalid or malformed `params_file` to ensure proper error handling.

Let me know if there are additional test cases you would like me to include!

saikishor, 2024-09-15:
> Hello @saikishor , I would like to contribute to this issue as my first step in open-source. Here is my plan for the tests:
> * Add test cases for passing a valid `params_file` and verifying that the [...]

Hello @SantoshGovindaraj!

Thank you for volunteering for this. I just gave a thought about this and it is way easier than I initially thought. 

What you can easily do is, in the test_spawner_unspawner.cpp you have the `cm` object. You can check the parameters `<controller_name>.params_file` parameter and see if it is the same as the unknown params file path in some of the tests and that should already take care of the required tests. 

Right now, the test_controller plugin doesn't have many parameters to test with a valid and invalid set, so the above option is better and easier.

I hope this helps. Thanks again.

SantoshGovindaraj, 2024-09-15:
Hi @saikishor , 
I realised that I missed sending you a review request. Unfortunately, I am unable to undo the request made by GitHub Action bot. Is there a way for me to send you a review request again?

Thank you!

saikishor, 2024-09-17:
Fixed with #1754

closed 2024-09-17 by saikishor