ros-controls/ros2_control #1704: Compatibility with ROS 2 Components

jcarpinelli-bdai, 2024-08-21:
### Summary

Can we use [`rclcpp_components`](https://github.com/ros2/rclcpp/tree/rolling/rclcpp_components), and [Components](https://docs.ros.org/en/humble/Concepts/Intermediate/About-Composition.html) more generally, to more effectively separate manager node / managed node semantics in the `ControllerManager` class?

### Issue

When using the `ControllerManager` class in a _new_ process (not the project-provided `ros2_control_node` process), renaming the controller manager might cause every underlying controller to have its private member variable node to _change its name_ to the name of the controller manager. This is a [known issue](https://github.com/ros-controls/ros2_control/issues/1684), and is described in the `ros2_control` [documentation](https://control.ros.org/humble/doc/ros2_control/controller_manager/doc/userdoc.html#using-the-controller-manager-in-a-process). 

### Composition

In ROS 2, Node [Composition](https://docs.ros.org/en/humble/Concepts/Intermediate/About-Composition.html) seems to be the general solution to the issue described above. Components can be [renamed](https://docs.ros.org/en/humble/Tutorials/Intermediate/Composition.html#remap-component-names-and-namespaces) independently of the component manager.

### Question

Does _composition_ (in the ROS 2 sense of the word) offer any benefits over the existing design of the controller manager, and the controller interface? Instead of inheriting a `std::shared_ptr<rclcpp_lifecycle::LifecycleNode>` in the controller interface, could controllers act as Node components? Could the controller manager be a component manager? Thank you for reading!

> [!WARNING]
>
> I am not a ROS 2 expert, so my understanding could definitely be incorrect. Please let me know if I'm misunderstanding `ros2_control`, or ROS 2 features more broadly.

saikishor, 2024-08-21:
Hello @jcarpinelli-bdai!

This is no longer a problem because the issue you are describing is solved with the https://github.com/ros-controls/ros2_control/pull/1694. The ros2 components that you are describing internally use the same approach.

Let us know if this fixes your issue.

Thank you!

jcarpinelli-bdai, 2024-08-21:
Thanks for the quick reply! In my own testing, setting `use_global_arguments(false)` does solve the re-naming issue, but it _also_ disables parameter setting for the composed nodes. I am using `humble`, so perhaps this behavior has changed in more recent ROS versions?

saikishor, 2024-08-21:
> Thanks for the quick reply! In my own testing, setting `use_global_arguments(false)` does solve the re-naming issue, but it _also_ disables parameter setting for the composed nodes. I am using `humble`, so perhaps this behavior has changed in more recent ROS versions?

Well that's true, that part will be affected. That's why the changes are not backported to older distros. However, in order to have it working use `--param-file` option from spawner, and this will help you set it to the controller nodes

christophfroehlich, 2024-08-22:
Thanks for the detailed summary of your issue. 
@saikishor Isn't the part in [docs](https://control.ros.org/humble/doc/ros2_control/controller_manager/doc/userdoc.html#using-the-controller-manager-in-a-process) outdated after #1694?

saikishor, 2024-08-22:
> Thanks for the detailed summary of your issue. @saikishor Isn't the part in [docs](https://control.ros.org/humble/doc/ros2_control/controller_manager/doc/userdoc.html#using-the-controller-manager-in-a-process) outdated after #1694?

Yes, it is outdated as now such individual node remapping is no longer needed, even though the same trick will still work. Should we cleanup this part of the docs?

jcarpinelli-bdai, 2024-08-22:
> Well that's true, that part will be affected. That's why the changes are not backported to older distros. However, in order to have it working use --param-file option from spawner, and this will help you set it to the controller nodes

Ah, I understand. Thank you for the clarification. I have been using global arguments passed to the controller manager node to configure each of the controllers, as shown in this [humble demo](https://github.com/ros-controls/ros2_control_demos/blob/f5a8fed5caf7db978e86b3941a53c910f1fee622/example_1/bringup/launch/rrbot.launch.py#L100). In demos for later releases, the `--params-file` argument is used instead. This all fully addresses my issue. Thanks again.

I am also interested, just to learn, how these "composed lifecycle nodes" used in `ros2_control` relate to `rclcpp_components`. If anyone is feeling like explaining... do components address the same needs as this pattern of "composing full lifecycle nodes"?