ros-controls/ros2_control #1714: Add tests for parsing ros arguments to the CM

saikishor, 2024-08-23:
We need to add tests parsing different --ros-args to the CM node to be able to propagate to the controllers. This will be needed to properly add testing to the changes in #1712

SantoshGovindaraj, 2024-09-22:
Hello @saikishor ,

I’d like to take on this issue and focus on adding tests for the function `ControllerManager::determine_controller_node_options`, where the `--ros-args` are processed and propagated to the controllers, I believe.

What I see that the function ensures:
1. Handling of `__ns` (namespace) and `__node` arguments to avoid conflicts.
2. Correct passing of parameter files through `--ros-args`.
3. Propagation of `use_sim_time` from the Controller Manager to the controllers.

So I plan to write tests covering:
- Node remapping (`__ns`, `__node`).
- Parameter file parsing.

Could you let me know if there are any additional test cases you'd like to include?

Thanks!

saikishor, 2024-09-22:
Hello @SantoshGovindaraj!

I don't think it would be needed. I have addressed testing of this in #1713, but I forgot to link the issue. Sorry for the confusion.

SantoshGovindaraj, 2024-09-22:
Thanks @saikishor for the update! I'll take a look at the tests in #1713. 
If I spot anything that could use more coverage, I'll let you know. Otherwise, I'll move on to another task.

PR #1713 (2024-12-19): Add controller node options args to be able to set controller specific node arguments
A different approach of #1712

```
    robot_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["diffbot_base_controller", "--param-file", controller_config, "--controller-ros-args", '--ros-args -r /diffbot_base_controller/cmd_vel:=/cmd_vel'],
    )
```

Fixes: #1711 
Fixes: #1714
closed 2024-12-19 by bmagyar