ros-controls/ros2_control #1716: hardware_interface does not work well when Isaac Sim is running

hijimasa, 2024-08-24:
**Describe the bug**
The command that is sent from controller is not sometimes reflected to hardware when Isaac Sim is running.

**To Reproduce**
Steps to reproduce the behavior:
1. launch Isaac Sim
2. launch ros2_control_demo_example_3 rrbot_system_multi_interface.launch.py
3. send topic by following command
   ```
   ros2 topic pub -1 /forward_velocity_controller/commands std_msgs/msg/Float64MultiArray "{layout: {dim: [], data_offset: 0}, data: [7.0, 7.0] }"
   or
   ros2 topic pub -1 /forward_velocity_controller/commands std_msgs/msg/Float64MultiArray "{layout: {dim: [], data_offset: 0}, data: [0.0, 0.0] }"
   ```
4. Sometimes the command that is sent from controller is not reflected to hardware

**Expected behavior**
All command that is sent from controller is reflected to hardware.

**Screenshots**
[Screencast from 2024年08月24日 01時45分17秒.webm](https://github.com/user-attachments/assets/57b05a8a-0a15-4495-ac9c-f183b9c7ffd8)

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS 2 Version: Humble
 - ros2_control Version: 2.42.0
 - Isaac Sim Version: 4.1

**Additional context**
The topic pub/sub works fine, so I assume it is a hardware_interface specific issue.
I have asked on Isaac Sim's forum, but I have also make this issue here, where there are contributors who are familiar with the hardware_interface implementation.

saikishor, 2024-08-24:
Hello @hijimasa!

Can you check this issue: https://github.com/ros-controls/ros2_controllers/issues/537 and see if it what you are facing?

Thank you!

hijimasa, 2024-08-24:
Hello @saikishor !

I already read it. And use_sim_time option is removed.
I think this is another issue.

saikishor, 2024-08-24:
> Hello @saikishor !
> 
> I already read it. And use_sim_time option is removed.
> I think this is another issue.

Hello @hijimasa!

The example 3 itself is working on its own. It's probably the environment you are using with Issac. Could you please recheck this?

Thank you!

hijimasa, 2024-08-24:
Hello @saikishor !

I know the example is working its own. I don't know how hardware_interface work. 
So I cannot understand where I should recheck in Isaac Sim.

hijimasa, 2024-08-25:
I changed ros2_controllers/forward_command_controller as below.
[forward_command_controller.zip](https://github.com/user-attachments/files/16738817/forward_command_controller.zip)

This change shows the command is lost between RealtimeBuffer read/write.
[Screencast from 2024年08月25日 11時51分12秒.webm](https://github.com/user-attachments/assets/19339c9e-8713-4659-b628-a98c781aba64)

I will try to check RealtimeBuffer.

hijimasa, 2024-08-25:
I changed ros2_controllers/forward_command_controller as below.
[forward_command_controller.zip](https://github.com/user-attachments/files/16738849/forward_command_controller.zip)

This change shows the command is lost between topic pub/sub, not RealtimeBuffer read/write.
[test.webm](https://github.com/user-attachments/assets/8a9aaad8-9506-4e2d-a87d-d041b82ef7a0)

I will try to check the controller QoS.

hijimasa, 2024-08-25:
This is topic pub/sub problem not ros2_control problem.
So I close this issue.

Thank you.

closed 2024-08-25 by hijimasa

saikishor, 2024-08-25:
Hello @hijimasa!

Could you please highlight the main change that you did?. It would be easier for future references. Did you find the problem already?

hijimasa, 2024-08-25:
Hello @saikishor !
Thank you for your reply.

I changed two points in src/forward_contollers_base.cpp.
1. on_configure() : I add RCLCPP_INFO() in subscriber callback function to check subscription.
2. update() : I add RCLCPP_INFO() to check RealtimeBuffer update.

I could not find the problem in QoS configuration but I think this is topic pub/sub problem not ros2_control problem.

saikishor, 2024-08-25:
It is indeed weird to have caused due to publisher and subscriber. If you get some lead on it. Let us know. It would help other users in future

hijimasa, 2024-08-25:
I don't know why Fast DDS doesn't work well, but Cyclone DDS solved the problem.
Isaac Sim and Fast DDS may be incompatible.

christophfroehlich, 2024-08-26:
One more reason to make cyclone the default RMW :see_no_evil:

saikishor, 2024-08-26:
> One more reason to make cyclone the default RMW 🙈

Agreed! DDS stuff is making the debugging much harder