ros-controls/ros2_control #1732: Allow diagnostics to be published from a hardware_interface

Timple, 2024-08-30:
**Is your feature request related to a problem? Please describe.**
For an actuator, often the actuator reports more values that just the measured one. For example current power consumed. It would be nice to have this information available on the `/diagnostics` topic.

**Describe the solution you'd like**
A way to easily supply diagnostics from a hardware_interface. For example:
```c++
hardware_interface::return_type MyActuator::read(const rclcpp::Time &, const rclcpp::Duration &);

hardware_interface::return_type MyActuator::write(const rclcpp::Time &, const rclcpp::Duration &);

hardware_interface::return_type MyActuator::produce_diagnostics(diagnostic_updater::DiagnosticStatusWrapper &stat);
```

Or:
```c++
hardware_interface::CallbackReturn on_configure(const rclcpp_lifecycle::State & previous_state);

hardware_interface::CallbackReturn add_diagnostic_task(const DiagnosticTask * t);
```

**Describe alternatives you've considered**
Creating a `diagnostic_updater_` class member. Which also needs a `node_` and then spin an executor. Like this: https://github.com/nicklgw/bdrive_hardware_interface/blob/f557bf1111f6ecc2df0389d25c7e73225e294a61/src/bdrive_hardware_interface.cpp#L93

**Additional context**

Timple, 2025-04-01:
Do any of the maintainers have a preferred way of implementing this?

soham2560, 2025-04-05:
@saikishor could you expand on why this won't work for system interfaces?

saikishor, 2025-04-05:
> [@saikishor](https://github.com/saikishor) could you expand on why this won't work for system interfaces with multiple joint?

Because one instance, can likely carry information of only one information level, and for the System we might have multiple actuators or sensors and if we need multi-level diagnostics or some flexibility, a single wrapper wouldn't be enough :). I hope this answers your question @soham2560