ros-controls/ros2_control #1753: Test `unload_on_kill_activate_as_group` fails on jazzy/testing

christophfroehlich, 2024-09-15:
```
2024-09-13T02:29:16.7694364Z 9: /home/runner/work/ros2_control/ros2_control/.work/target_ws/src/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:347: Failure
2024-09-13T02:29:16.7694540Z 9: Expected equality of these values:
2024-09-13T02:29:16.7694700Z 9:   cm_->get_loaded_controllers().size()
2024-09-13T02:29:16.7694790Z 9:     Which is: 2
2024-09-13T02:29:16.7694871Z 9:   0ul
2024-09-13T02:29:16.7694961Z 9:     Which is: 0
2024-09-13T02:29:16.7704118Z 9: 
```
see https://github.com/ros-controls/ros2_control/issues/1751

christophfroehlich, 2024-09-15:
@saikishor could you have a look maybe?

saikishor, 2024-09-15:
@christophfroehlich Is it failing more frequently?

saikishor, 2024-09-15:
```
     [INFO] [1726194509.916911268] [spawner_ctrl_3]: [NON-XML-CHAR-0x1B][92mConfigured and activated all the parsed controllers list : ['ctrl_3', 'ctrl_2']![NON-XML-CHAR-0x1B][0m
    [INFO] [1726194509.917833389] [spawner_ctrl_3]: Waiting until interrupt to unload controllers
    /home/runner/work/ros2_control/ros2_control/.work/target_ws/src/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:347: Failure
    Expected equality of these values:
      cm_->get_loaded_controllers().size()
        Which is: 2
      0ul
        Which is: 0
```

Seems like it didn't catch the keyboard interrupt. Let's wait to see if this happens more often.

Thank you!

saikishor, 2024-09-16:
> Now and then, but at the next try it succeeded

Gotcha! We should keep an eye on this.

christophfroehlich, 2024-10-08:
https://github.com/ros-controls/ros2_control_ci/actions/runs/11227370628/job/31209485522

christophfroehlich, 2025-01-02:
https://github.com/ros-controls/ros2_control_ci/actions/runs/12554458549

christophfroehlich, 2025-03-07:
https://github.com/ros-controls/ros2_control_ci/issues/267

christophfroehlich, 2025-04-30:
#2211

christophfroehlich, 2025-05-17:
https://github.com/ros-controls/ros2_control/issues/2238

christophfroehlich, 2025-05-27:
https://github.com/ros-controls/ros2_control_ci/issues/342

christophfroehlich, 2025-06-09:
#2322

christophfroehlich, 2025-06-09:
#2320

christophfroehlich, 2025-06-11:
https://github.com/ros-controls/ros2_control/issues/2325

saikishor, 2025-07-15:
```
   Traceback (most recent call last):
      File "/home/runner/work/ros2_control/ros2_control/.work/target_ws/install/controller_manager/lib/python3.12/site-packages/controller_manager/spawner.py", line 352, in main
        time.sleep(1)
    KeyboardInterrupt
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<frozen runpy>", line 198, in _run_module_as_main
      File "<frozen runpy>", line 88, in _run_code
      File "/usr/lib/python3/dist-packages/coverage/__main__.py", line 10, in <module>
        sys.exit(main())
                 ^^^^^^
      File "/usr/lib/python3/dist-packages/coverage/cmdline.py", line 970, in main
        status = CoverageScript().command_line(argv)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/usr/lib/python3/dist-packages/coverage/cmdline.py", line 681, in command_line
        return self.do_run(options, args)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/usr/lib/python3/dist-packages/coverage/cmdline.py", line 858, in do_run
        runner.run()
      File "/usr/lib/python3/dist-packages/coverage/execfile.py", line 211, in run
        exec(code, main_mod.__dict__)
      File "/home/runner/work/ros2_control/ros2_control/.work/target_ws/install/controller_manager/lib/controller_manager/spawner", line 33, in <module>
        sys.exit(load_entry_point('controller-manager==4.33.0', 'console_scripts', 'spawner')())
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/runner/work/ros2_control/ros2_control/.work/target_ws/install/controller_manager/lib/python3.12/site-packages/controller_manager/spawner.py", line 355, in main
        logger.info("KeyboardInterrupt successfully captured!")
      File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/impl/rcutils_logger.py", line 341, in info
        def info(self, message, **kwargs):
    
    KeyboardInterrupt
    /home/runner/work/ros2_control/ros2_control/.work/target_ws/src/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:601: Failure
    Expected equality of these values:
      cm_->get_loaded_controllers().size()
        Which is: 2
      0ul
        Which is: 0
    
    [INFO] [1752550945.715375762] [test_controller_manager.pal_statistics]: Async messages lost 0
    [INFO] [1752550945.715406460] [test_controller_manager.pal_statistics]: publish_async_failures_ 0
```

It seems like the second exception is partly from the coverage script

christophfroehlich, 2025-08-04:
https://github.com/ros-controls/ros2_control/issues/2438

christophfroehlich, 2025-11-01:
https://github.com/ros-controls/ros2_control/actions/runs/18991010496/job/54243979686?pr=2765
jazzy/RHEL

```
     [  FAILED  ] TestLoadController.unload_on_kill_activate_as_group
    
     1 FAILED TEST
    -- run_test.py: return code 1
    -- run_test.py: inject classname prefix into gtest result file '/__w/ros2_control/ros2_control/build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml'
    -- run_test.py: verify result file '/__w/ros2_control/ros2_control/build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml'
  >>>
build/controller_manager/test_results/controller_manager/test_spawner_unspawner.gtest.xml: 26 tests, 0 errors, 1 failure, 0 skipped
- controller_manager.TestLoadController unload_on_kill_activate_as_group (/__w/ros2_control/ros2_control/src/ros-controls/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:585)
  <<< failure message
    /__w/ros2_control/ros2_control/src/ros-controls/ros2_control/controller_manager/test/test_spawner_unspawner.cpp:601
    Expected equality of these values:
      cm_->get_loaded_controllers().size()
        Which is: 2
      0ul
        Which is: 0
  >>>
```

christophfroehlich, 2025-12-15:
https://github.com/ros-controls/ros2_control/issues/2901

christophfroehlich, 2026-03-04:
https://github.com/ros-controls/ros2_control/issues/3074