ros-controls/ros2_control #1757: Warn user when a resource is taking a long time to initialize

EzraBrooks, 2024-09-18:
**Is your feature request related to a problem? Please describe.**
When developing a custom robot, our driver was freezing when connecting to a particular motor. This resulted in the `/controller_manager` never advertising the services, because the resource manager never marked the components as initialized.

**Describe the solution you'd like**
We eventually figured out that it was our fault, but it would've been very useful to have the resource manager log a warning indicating that the resources had taken a really long time to start.

saikishor, 2024-09-19:
Hello @EzraBrooks!

There are some logs printed in debug mode. Right here : https://github.com/ros-controls/ros2_control/blob/master/hardware_interface%2Fsrc%2Fsystem.cpp#L269-L275. Does this help?

The idea is that each hardware takes its own time to initialize and activate, so most of these are decoupled from read and write cycles.

christophfroehlich, 2024-09-19:
Could we add a timeout for the initialization? And add it as a parameter to the [hardware_spawner](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#hardware-spawner)?

saikishor, 2024-09-19:
> Could we add a timeout for the initialization? And add it as a parameter to the [hardware_spawner](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#hardware-spawner)?

I don't know if we have control over the initialisation time, as we don't spawn in a different thread to account for the time. Could be, but I'm not sure

EzraBrooks, 2024-09-19:
If I run into the problem again today, I can try bumping the log level to debug and see if those logs are useful, but this is my last day with this particular robot and I need to get some other stuff working, so no promises on that front :/

saikishor, 2024-09-19:
> If I run into the problem again today, I can try bumping the log level to debug and see if those logs are useful, but this is my last day hands-on with this particular robot and I need to get some other stuff working, so no promises on that front :/

Sure. Thank you!

EzraBrooks, 2024-09-19:
unfortunately I wasn't able to get around to that test. I do like @christophfroehlich's idea if possible, though