ros-controls/ros2_control #1761: [spawner_joint_trajectory_controller]: Could not contact service /controller_manager/list_controllers (UR robot)

LucaMIMOUNI, 2024-09-23:
**Describe the bug**
I am trying to launch the simulation for a UR10e robot using [Universal_Robots_ROS2_Gazebo_Simulation](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation). When launching `ros2 launch ur_simulation_gazebo ur_sim_moveit.launch.py`, I get the following error: 
[spawner-2] [INFO] [1727077744.034227709] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-3] [WARN] [1727077744.081321269] [spawner_joint_trajectory_controller]: Could not contact service /controller_manager/list_controllers

(Cf screenshots below for the full message error)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [Universal_Robots_ROS2_Gazebo_Simulation](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation) and [Universal_Robots_ROS2_Description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description) to get both simulation launch file and description of the robot.
2. git clone repertories in your workspace
3. do `ros2 launch ur_simulation_gazebo ur_sim_moveit.launch.py`
4. See error

**Behavior**

- Expected to launch correctly the controller in order to start move the robot. 
- Gazebo simulation takes lot of times to load the environment and finally load the following (See 1st screenshot). 
- Rviz seems to load correctly, and, thanks to moveit!, successfully plan a trajectory, but of course can't execute it (because controller_manager not available). (Cf 2nd screenshot)

**Screenshots**
[image]

[image]

[image]

[image]

[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04 Jammy
 - Version ROS Humble
- Gazebo 11.10.2

**Additional context**
I have install humble through deb packages, but got similar error with source. I have install correctly ros2-control and ros2-controllers through `sudo apt install ros-humble-ros2-control` and `sudo apt install ros-humble-ros2-controllers`.

Finally, I've tried to launch following example: [ros2_control_demo_example_9_Gazebo](https://control.ros.org/humble/doc/ros2_control_demos/example_9/doc/userdoc.html) and controller_manager load well ! But when doing `ros2 launch ros2_control_demo_example_9 test_forward_position_controller.launch.py` or simply `ros2 topic pub /forward_position_controller/commands std_msgs/msg/Float64MultiArray "data:- 0.5 - 0.5"`, robot start to teleport directly to the desired position, so here (0.5, 0.5).
Thanks !

fmauch, 2024-09-23:
I just tried this locally and cannot really reproduce that. It sounds a bit like the problem, we tried to solve in #1680. Which version of ros2_control do you currently have installed? The fix should be available from version 2.43 on.

LucaMIMOUNI, 2024-09-23:
As you see below, it should be newest version for both ros control and ros controllers:

[image]
[image]

I m trying to understand what you did in #1680 , should I pull something ?

fmauch, 2024-09-23:
> I m trying to understand what you did in #1680 , should I pull something ?

No, that should already be included in 2.43.0.

Looking at your output in more detail (Please use a [code section ](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks) in future posts, that is much easier to parse than images) I am missing any output of `gazebo_ros2_control`. Do you have the package `ros-humble-gazebo-ros2-control` installed?

I would expect some kind of error output, though, if you did not have that installed.

Specifically, I am missing something like this:

```
[gzserver-4] [INFO] [1727089566.261520935] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin                                                                                                                                                                                                                        
[gzserver-4] [INFO] [1727089566.307508601] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /                                                                                                                                                                                                       
[gzserver-4] [INFO] [1727089566.308089400] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control                                                                                                                                                                                    
[gzserver-4] [INFO] [1727089566.357301674] [gazebo_ros2_control]: connected to service!! robot_state_publisher                                                                                                                                                                                                              
[gzserver-4] [INFO] [1727089566.369991680] [gazebo_ros2_control]: Received urdf from param server, parsing...                                                                                                                                                                                                               
[gzserver-4] [INFO] [1727089566.370167800] [gazebo_ros2_control]: Loading parameter files /home/ubuntu/checkout/ur_humble/colcon_ws/install/ur_simulation_gazebo/share/ur_simulation_gazebo/config/ur_controllers.yaml                                                                                                      
[gzserver-4] [INFO] [1727089566.497004393] [gazebo_ros2_control]: Loading joint: shoulder_pan_joint                                                                                                                                                                                                                         
[gzserver-4] [INFO] [1727089566.497099555] [gazebo_ros2_control]:       State:                                                                                                                                                                                                                                              
[gzserver-4] [INFO] [1727089566.497119370] [gazebo_ros2_control]:                position                                                                                                                                                                                                                                   
[gzserver-4] [INFO] [1727089566.497193085] [gazebo_ros2_control]:                        found initial value: 0.000000                                                                                                                                                                                                      
[gzserver-4] [INFO] [1727089566.497211993] [gazebo_ros2_control]:                velocity                                                                                                                                                                                                                                   
[gzserver-4] [INFO] [1727089566.497222375] [gazebo_ros2_control]:                effort                                                                                                                                                                                                                                     
[gzserver-4] [INFO] [1727089566.497250768] [gazebo_ros2_control]:       Command:                                                                              
[gzserver-4] [INFO] [1727089566.497261095] [gazebo_ros2_control]:                position                                                                     
[gzserver-4] [INFO] [1727089566.501875324] [gazebo_ros2_control]:                velocity                                                                     
[gzserver-4] [INFO] [1727089566.502011320] [gazebo_ros2_control]: Loading joint: shoulder_lift_joint                                                          
[gzserver-4] [INFO] [1727089566.502032088] [gazebo_ros2_control]:       State:                                                                                
[gzserver-4] [INFO] [1727089566.502043524] [gazebo_ros2_control]:                position                                                                     
[... cut ...]
[gzserver-4] [INFO] [1727089566.503952590] [resource_manager]: 'activate' hardware 'ur' 
[gzserver-4] [INFO] [1727089566.503961913] [resource_manager]: Successful 'activate' of hardware 'ur'
[gzserver-4] [INFO] [1727089566.509357677] [gazebo_ros2_control]: Loading controller_manager                          
[gzserver-4] [WARN] [1727089566.705769938] [gazebo_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-4] [INFO] [1727089566.707405797] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
```

LucaMIMOUNI, 2024-09-23:
Here is my output with codesection (Didn't know about this thx):

```
luca@luca-TravelMate-TCO:~$ ros2 launch ur_simulation_gazebo ur_sim_moveit.launch.py
[INFO] [launch]: All log files can be found below /home/luca/.ros/log/2024-09-23-13-23-16-847606-luca-TravelMate-TCO-77010
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [77021]
[INFO] [spawner-2]: process started with pid [77023]
[INFO] [spawner-3]: process started with pid [77025]
[INFO] [gzserver-4]: process started with pid [77027]
[INFO] [gzclient-5]: process started with pid [77029]
[INFO] [spawn_entity.py-6]: process started with pid [77031]
[INFO] [move_group-7]: process started with pid [77033]
[INFO] [rviz2-8]: process started with pid [77035]
[INFO] [servo_node_main-9]: process started with pid [77037]
[rviz2-8] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[servo_node_main-9] [WARN] [1727090597.860677814] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-9] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-9] to the Servo composable node in the launch file
[move_group-7] [WARN] [1727090597.869557410] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[robot_state_publisher-1] [INFO] [1727090597.883865970] [robot_state_publisher]: got segment base
[robot_state_publisher-1] [INFO] [1727090597.883985471] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1727090597.884000456] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-1] [INFO] [1727090597.884010461] [robot_state_publisher]: got segment flange
[robot_state_publisher-1] [INFO] [1727090597.884019230] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-1] [INFO] [1727090597.884027667] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-1] [INFO] [1727090597.884035769] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-1] [INFO] [1727090597.884043754] [robot_state_publisher]: got segment tool0
[robot_state_publisher-1] [INFO] [1727090597.884052269] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-1] [INFO] [1727090597.884060419] [robot_state_publisher]: got segment world
[robot_state_publisher-1] [INFO] [1727090597.884068565] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-1] [INFO] [1727090597.884076457] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-1] [INFO] [1727090597.884084510] [robot_state_publisher]: got segment wrist_3_link
[servo_node_main-9] [INFO] [1727090597.888011392] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.026155 seconds
[servo_node_main-9] [INFO] [1727090597.888053554] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-9] [INFO] [1727090597.888063315] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-9] [WARN] [1727090597.900488665] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[move_group-7] [INFO] [1727090597.903603989] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-7] [INFO] [1727090597.903641225] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-7] [INFO] [1727090597.903650023] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-9] [INFO] [1727090597.934488986] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[servo_node_main-9] [INFO] [1727090597.937409553] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[servo_node_main-9] [INFO] [1727090597.937441869] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[servo_node_main-9] [INFO] [1727090597.938187317] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[servo_node_main-9] [INFO] [1727090597.939476733] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on '/servo_node/publish_planning_scene'
[move_group-7] [INFO] [1727090597.957689770] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-7] [INFO] [1727090597.957886505] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-7] [INFO] [1727090597.958280220] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-7] [INFO] [1727090597.958673277] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-7] [INFO] [1727090597.958692394] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-7] [INFO] [1727090597.960101368] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[... cut ...]
Could maybe come from my bash.rc. I tried to use a different dds, but don't changed anything. Also have both humble and gazebo_ros2_control from source, but don't seems to change anything too, so I m using debian packages installed with command: `sudo apt install <debian_pkg>`
```
# DDS
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

# Sourcing Ros2
#source /home/luca/ros2_humble/install/setup.bash
source /opt/ros/humble/setup.bash
source /home/luca/ros2_ws/install/setup.bash

# Sourcing gazebo
source /usr/share/gazebo/setup.sh
#source /home/luca/gz_ros2_control_ws/install/setup.bash
```

LucaMIMOUNI, 2024-09-23:
Just find the culprit ! When going on this link [Ur_Gazebo_HUMBLE](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/tree/humble) and git clone it, it was git cloning the rolling branch !

The correct command line is `git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Description.git`, and now it's working fine. Thats totally my bad, as a beginner I didn't notice that I wasn't git clonning the humble branch even if I was on the correct web link [Ur_Gazebo_HUMBLE](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/tree/humble).

Thanks sir to have taken bit of your time to help me.

closed 2024-09-23 by LucaMIMOUNI

fmauch, 2024-09-24:
Ah, that totally explains it. For rolling we refactored things so that the ros2_control tag gets added in the ur_simulation_gazebo package. So, you never had a ros2_control tag and hence never had any gazebo_ros2_control started.

I'm glad that things are working for you now!