ros-controls/ros2_control #1795: [Iron] Move deprecation warning message "class_type" into separate logger

firesurfer, 2024-10-15:
# Description

When we call ` ros2 service call /controller_manager/list_hardware_components controller_manager_msgs/srv/ListHardwareComponents` in iron the controller manager prints 6 times:

```
[ros2_control_node-19] [WARN] [1728976987.424427528] [controller_manager]: The 'class_type' field in 'controller_manager_msgs/msg/HardwareComponentState.msg' is deprecated and will be removed soon. Please switch over client code to use 'plugin_name' instead.
```

As I have parts in the system that automatically call this service once in a while it really spams the log. 

# Action

Please move the warning in a sublogger: for example `controller_manager.deprecation` that can be silenced separately. 

It should then be possible to silence the specific logger via command line / launch argument:

From the docs:

```
ros2 run logging_demo logging_demo_main --ros-args --log-level logger_usage_demo:=debug
```

saikishor, 2024-10-15:
@firesurfer feel free to open a PR with your suggestion and let the maintainers review it

christophfroehlich, 2025-01-22:
I think this is obsolete cause iron has reached EOL.

closed 2025-01-22 by christophfroehlich