ros-controls/ros2_control #1802: Issue: Need Help with Bosch BNO055 Sensor Example

racarla96, 2024-10-18:
Hi there,

I’m trying to work with the Bosch BNO055 sensor example in the [ros2_caddy_ai2_sensors_Bosch_BNO055](https://github.com/racarla96/ros2_caddy_ai2_sensors_Bosch_BNO055) repository, but I'm having some trouble getting it to work. I’m new to this project and open source in general, and I’m not sure how to proceed.

I was trying to follow the example to build a package to run just one Sensor Interface, but I don't know how to proceed....
Right now the actual sensor logic is not implemented...

Thank you in advance for your help!

christophfroehlich, 2024-10-18:
The link is not working (the repo is private?). 

Have you seen the [demos we maintain](https://control.ros.org/rolling/doc/ros2_control_demos/doc/index.html)? Find also a step-by-step guide for writing your hardware component [here](https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html). 

Apart from that we can't give you support in implementing your hardware. You could ask on https://robotics.stackexchange.com/ if someone has written already a ROS driver for that, or you can look for professional services to do the work for you.

closed 2024-10-18 by christophfroehlich

racarla96, 2024-10-21:
Thank you!

Yes, it's private, sorry. Now, it's public.

I've re-read the documentation, I think the next step would be to make a controller that uses this hardware interface, from what I understand.

christophfroehlich, 2024-10-25:
That depends what you want to do with the sensor. You could also use the [imu_sensor semantic component](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/semantic_components/imu_sensor.hpp) and use the imu_sensor_broadcaster -> then you don't need to write your own broadcaster (aka controller).

bmagyar, 2024-10-25:
Or you write your controller for taking that sensor into account for doing
something with your robot.