ros-controls/ros2_control #1806: unconfigured problem

naorwaiss, 2024-10-20:
hi every one i am new here and try at the first time to configured my gazebo robot with ros2 control 
the problem is the controller is unconfigure and i dont find where the problem:

the urdf file 
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro" name= "x8_robot">
 
<xacro:arg name="is_ignition" default="true"/>
<!--    links-->

<link name="base_link"></link>
<link name="chassis"><inertial><origin xyz="-0.00984195359459856 -0.00120925257982298 0.0505150963957082" rpy="0 0 0" /><mass value="3.83579845094888" /><inertia
    ixx="0.0108641330207659"
    ixy="-5.9823674702839E-07"
    ixz="-1.768908869104E-06"
    iyy="0.0199873663409641"
    iyz="2.49003604337839E-05"
    izz="0.0251403002191233"
/></inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/base_link.STL"  /></geometry><material name=""><color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/base_link.STL" /></geometry></collision></link>

<link name="front_left">
<inertial><origin xyz="-0.151392376561602 -0.0765244450560608 0.045198369230977" rpy="0 0 0" /><mass value="0.729027057768021" /><inertia
    ixx="0.00137456304544284"
    ixy="-0.00131718035914277"
    ixz="-7.24913767818369E-05"
    iyy="0.00280935567626702"
    iyz="-5.18206845156667E-05"
    izz="0.00402554483982116"
/></inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/front_left.STL" /></geometry><material name=""><color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/front_left.STL" /></geometry></collision></link>

<link name="front_front_left"><inertial><origin xyz="2.7273E-08 8.4788E-07 0.033785" rpy="0 0 0" /><mass value="0.62345" /><inertia
  ixx="0.0021571"
  ixy="5.7178E-10"
  ixz="1.3794E-10"
  iyy="0.0021571"
  iyz="1.1818E-08"
  izz="0.0038015"
/></inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry><material name=""><color rgba="0.79216 0.81961 0.93333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry></collision></link>

<link name="front_back_left"><inertial><origin xyz="2.7284E-08 8.4791E-07 0.033785" rpy="0 0 0" /><mass value="0.62345" /><inertia
  ixx="0.0021571"
  ixy="5.7162E-10"
  ixz="1.3798E-10"
  iyy="0.0021571"
  iyz="1.1818E-08"
  izz="0.0038015"
/></inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry><material name=""><color rgba="0.79216 0.81961 0.93333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry></collision></link>

 <link name="front_right"><inertial><origin xyz="-0.15140299531934 -0.076526205656618 -0.0451940468029265" rpy="0 0 0" /><mass value="0.729214138297053"/><inertia
  ixx="0.00137478227627483"
  ixy="-0.00131726895128445"
  ixz="7.25804426030585E-05"
  iyy="0.00281017626017176"
  iyz="5.18669798425697E-05"
  izz="0.00402653973335046" />
</inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/front_right.STL" /></geometry><material name=""><color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/front_right.STL" /></geometry></collision></link>

<link name="front_back_right"><inertial><origin xyz="2.7284E-08 8.4791E-07 0.033785" rpy="0 0 0" /><mass value="0.62345" /><inertia
  ixx="0.0021571"
  ixy="5.7162E-10"
  ixz="1.3798E-10"
  iyy="0.0021571"
  iyz="1.1818E-08"
  izz="0.0038015"
  /></inertial><visual><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry><material name=""><color rgba="0.79216 0.81961 0.93333 1" /></material></visual><collision><origin xyz="0 0 0" rpy="0 0 0" /><geometry><mesh filename="package://x8_sim/meshes/wheel.STL" /></geometry></collision></link>

<link name="front_front_right"><inertial><origin xyz="2.7273E-08 8.4788E-07 0.033785" rpy="0 0 0" /><mass value="0.62345" /><inertia
    ixx="0.0021571"
    ixy="5.7178E-10"
[... 381 lines cut ...]
          - front_left_joint
          - front_right_joint
        command_interfaces:
          - position
        state_interfaces:
          - position
          - velocity

    velocity_controller:
      ros__parameters:
        joints:
          - back_back_left_joint
          - back_back_right_joint
          - front_front_left_joint
          - front_front_right_joint
        command_interfaces:
          - velocity
        state_interfaces:
          - position
          - velocity

```

all the partf of the gzebo work but i didnt understand the  problem of the unconfigured 
thanks for the help

christophfroehlich, 2024-10-20:
please add the console output from both launch files, or we can't help you.

naorwaiss, 2024-10-21:
ok 
so the first launch file - the gazebo one
```
ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/Desktop/naor/work/x8_c++$ ros2 launch x8_sim robot_gazebo_launch.py 
[INFO] [launch]: All log files can be found below /home/naor/.ros/log/2024-10-21-14-02-22-432865-naor-ASUS-TUF-Gaming-F15-FX507VV-FX507VV-133403
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [133404]
[INFO] [gzclient-2]: process started with pid [133406]
[INFO] [robot_state_publisher-3]: process started with pid [133408]
[robot_state_publisher-3] [INFO] [1729508542.938452285] [robot_state_publisher]: got segment back_back_left
[robot_state_publisher-3] [INFO] [1729508542.938522009] [robot_state_publisher]: got segment back_back_right
[robot_state_publisher-3] [INFO] [1729508542.938527323] [robot_state_publisher]: got segment back_front_left
[robot_state_publisher-3] [INFO] [1729508542.938530274] [robot_state_publisher]: got segment back_front_right
[robot_state_publisher-3] [INFO] [1729508542.938533389] [robot_state_publisher]: got segment back_left
[robot_state_publisher-3] [INFO] [1729508542.938536096] [robot_state_publisher]: got segment back_right
[robot_state_publisher-3] [INFO] [1729508542.938538807] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1729508542.938541497] [robot_state_publisher]: got segment chassis
[robot_state_publisher-3] [INFO] [1729508542.938544111] [robot_state_publisher]: got segment first_joint
[robot_state_publisher-3] [INFO] [1729508542.938546832] [robot_state_publisher]: got segment front_back_left
[robot_state_publisher-3] [INFO] [1729508542.938549454] [robot_state_publisher]: got segment front_back_right
[robot_state_publisher-3] [INFO] [1729508542.938552125] [robot_state_publisher]: got segment front_front_left
[robot_state_publisher-3] [INFO] [1729508542.938554666] [robot_state_publisher]: got segment front_front_right
[robot_state_publisher-3] [INFO] [1729508542.938557399] [robot_state_publisher]: got segment front_left
[robot_state_publisher-3] [INFO] [1729508542.938559988] [robot_state_publisher]: got segment front_right
[robot_state_publisher-3] [INFO] [1729508542.938562695] [robot_state_publisher]: got segment secend_joint
[robot_state_publisher-3] [INFO] [1729508542.938565447] [robot_state_publisher]: got segment third_joint
[INFO] [spawn_entity.py-4]: process started with pid [133620]
[spawn_entity.py-4] [INFO] [1729508548.230213235] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1729508548.230471288] [spawn_entity]: Loading entity XML from file /tmp/tmp43dwaagn.urdf
[spawn_entity.py-4] [INFO] [1729508548.231268255] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1729508548.231469206] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1729508548.234092137] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1729508548.442455116] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [x8_robot]
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 133620]
[gzserver-1] [INFO] [1729508552.745815449] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1729508552.748285404] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1729508552.748517256] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1729508552.750214586] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1729508552.750779725] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1729508552.750810170] [gazebo_ros2_control]: Loading parameter files /home/naor/Desktop/naor/work/x8_c++/src/x8_sim/config/my_controllers.yaml
[gzserver-1] [INFO] [1729508552.759241529] [gazebo_ros2_control]: Loading joint: front_left_joint
[gzserver-1] [INFO] [1729508552.759306482] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729508552.759313250] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729508552.759320845] [gazebo_ros2_control]:                        found initial value: 0.000000
[gzserver-1] [INFO] [1729508552.759329875] [gazebo_ros2_control]:                velocity
[gzserver-1] [INFO] [1729508552.759333516] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1729508552.759336783] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729508552.759370205] [gazebo_ros2_control]: Loading joint: front_right_joint
[gzserver-1] [INFO] [1729508552.759374399] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729508552.759377392] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729508552.759380442] [gazebo_ros2_control]:                        found initial value: 0.000000
[gzserver-1] [INFO] [1729508552.759383917] [gazebo_ros2_control]:                velocity
[gzserver-1] [INFO] [1729508552.759386826] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1729508552.759390085] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729508552.759398855] [gazebo_ros2_control]: Loading joint: back_right_joint
[gzserver-1] [INFO] [1729508552.759402128] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729508552.759405166] [gazebo_ros2_control]:                position
[... cut ...]
[spawner-1] [INFO] [1729508606.692269460] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[ERROR] [spawner-2]: process has died [pid 133861, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner position_controller --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-3]: process has died [pid 133863, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner velocity_controller --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-1]: process has finished cleanly [pid 133859]
```

christophfroehlich, 2024-10-21:
Isn't there any additional output in the gazebo launch file after you launch the controller one?

naorwaiss, 2024-10-21:
umm this is the the gazebo after the launch of the controller 

```
naor@naor-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/Desktop/naor/work/x8_c++$ ros2 launch x8_sim robot_gazebo_launch.py 
[INFO] [launch]: All log files can be found below /home/naor/.ros/log/2024-10-21-16-37-59-463960-naor-ASUS-TUF-Gaming-F15-FX507VV-FX507VV-9544
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [9545]
[INFO] [gzclient-2]: process started with pid [9547]
[INFO] [robot_state_publisher-3]: process started with pid [9549]
[robot_state_publisher-3] [INFO] [1729517880.025390102] [robot_state_publisher]: got segment back_back_left
[robot_state_publisher-3] [INFO] [1729517880.025471489] [robot_state_publisher]: got segment back_back_right
[robot_state_publisher-3] [INFO] [1729517880.025478050] [robot_state_publisher]: got segment back_front_left
[robot_state_publisher-3] [INFO] [1729517880.025481602] [robot_state_publisher]: got segment back_front_right
[robot_state_publisher-3] [INFO] [1729517880.025484873] [robot_state_publisher]: got segment back_left
[robot_state_publisher-3] [INFO] [1729517880.025488129] [robot_state_publisher]: got segment back_right
[robot_state_publisher-3] [INFO] [1729517880.025491229] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1729517880.025494583] [robot_state_publisher]: got segment chassis
[robot_state_publisher-3] [INFO] [1729517880.025497734] [robot_state_publisher]: got segment first_joint
[robot_state_publisher-3] [INFO] [1729517880.025500743] [robot_state_publisher]: got segment front_back_left
[robot_state_publisher-3] [INFO] [1729517880.025503820] [robot_state_publisher]: got segment front_back_right
[robot_state_publisher-3] [INFO] [1729517880.025506845] [robot_state_publisher]: got segment front_front_left
[robot_state_publisher-3] [INFO] [1729517880.025509881] [robot_state_publisher]: got segment front_front_right
[robot_state_publisher-3] [INFO] [1729517880.025512870] [robot_state_publisher]: got segment front_left
[robot_state_publisher-3] [INFO] [1729517880.025520998] [robot_state_publisher]: got segment front_right
[robot_state_publisher-3] [INFO] [1729517880.025525380] [robot_state_publisher]: got segment secend_joint
[robot_state_publisher-3] [INFO] [1729517880.025528525] [robot_state_publisher]: got segment third_joint
[INFO] [spawn_entity.py-4]: process started with pid [9792]
[spawn_entity.py-4] [INFO] [1729517885.639993097] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1729517885.640192442] [spawn_entity]: Loading entity XML from file /tmp/tmpzq4skibo.urdf
[spawn_entity.py-4] [INFO] [1729517885.641882440] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1729517885.642067021] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1729517885.646910953] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1729517885.942311731] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [x8_robot]
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 9792]
[gzserver-1] [INFO] [1729517890.326004828] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1729517890.328329742] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /
[gzserver-1] [INFO] [1729517890.328534164] [gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1729517890.329764317] [gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1729517890.330322988] [gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1729517890.330345722] [gazebo_ros2_control]: Loading parameter files /home/naor/Desktop/naor/work/x8_c++/src/x8_sim/config/my_controllers.yaml
[gzserver-1] [INFO] [1729517890.340875227] [gazebo_ros2_control]: Loading joint: front_left_joint
[gzserver-1] [INFO] [1729517890.340951735] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729517890.340959835] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729517890.340968227] [gazebo_ros2_control]:                        found initial value: 0.000000
[gzserver-1] [INFO] [1729517890.340975445] [gazebo_ros2_control]:                velocity
[gzserver-1] [INFO] [1729517890.340979227] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1729517890.340982389] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729517890.341192074] [gazebo_ros2_control]: Loading joint: front_right_joint
[gzserver-1] [INFO] [1729517890.341196584] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729517890.341199804] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729517890.341203318] [gazebo_ros2_control]:                        found initial value: 0.000000
[gzserver-1] [INFO] [1729517890.341206794] [gazebo_ros2_control]:                velocity
[gzserver-1] [INFO] [1729517890.341209917] [gazebo_ros2_control]:       Command:
[gzserver-1] [INFO] [1729517890.341212951] [gazebo_ros2_control]:                position
[gzserver-1] [INFO] [1729517890.341227727] [gazebo_ros2_control]: Loading joint: back_right_joint
[gzserver-1] [INFO] [1729517890.341230895] [gazebo_ros2_control]:       State:
[gzserver-1] [INFO] [1729517890.341233973] [gazebo_ros2_control]:                position
[... cut ...]
[gzserver-1] [ERROR] [1729517890.661875180] [controller_manager]: After configuring, controller 'velocity_controller' is in state 'unconfigured' , expected inactive.
[gzserver-1] [INFO] [1729517890.661944977] [controller_manager]: Configuring controller 'position_controller'
[gzserver-1] [ERROR] [1729517890.661958298] [position_controller]: 'joints' parameter was empty
[gzserver-1] [WARN] [1729517890.661967130] [position_controller]: Error occurred while doing error handling.
[gzserver-1] [ERROR] [1729517890.661972750] [controller_manager]: After configuring, controller 'position_controller' is in state 'unconfigured' , expected inactive.

```

naorwaiss, 2024-10-24:
??

saikishor, 2024-10-24:
Hello @naorwaiss!

It seems there is an issue in your configuration. The joints param and some parameters are not set at all, that's why the controller is failing.

```
[gzserver-1] [INFO] [1729517890.661671672] [controller_manager]: Configuring controller 'velocity_controller'
[gzserver-1] [ERROR] [1729517890.661791468] [velocity_controller]: 'joints' parameter was empty
[gzserver-1] [WARN] [1729517890.661829242] [velocity_controller]: Error occurred while doing error handling.
[gzserver-1] [ERROR] [1729517890.661875180] [controller_manager]: After configuring, controller 'velocity_controller' is in state 'unconfigured' , expected inactive.
[gzserver-1] [INFO] [1729517890.661944977] [controller_manager]: Configuring controller 'position_controller'
[gzserver-1] [ERROR] [1729517890.661958298] [position_controller]: 'joints' parameter was empty
[gzserver-1] [WARN] [1729517890.661967130] [position_controller]: Error occurred while doing error handling.
[gzserver-1] [ERROR] [1729517890.661972750] [controller_manager]: After configuring, controller 'position_controller' is in state 'unconfigured' , expected inactive
```

naorwaiss, 2024-10-24:
you can say what worng with my config ?
and what i need to config - say the topic and i do the research..

saikishor, 2024-10-24:
> you can say what worng with my config ? and what i need to config - say the topic and i do the research..

Your YAML configuration is wrong, you need to have the  controller things outside the controller_manager namespace and then  pass this file to be loaded by the spawner using `--params-file` or `-p` option

christophfroehlich, 2024-10-25:
It seems that this issue can be closed.

For such types of questions (seeking help in setting things up), please use https://robotics.stackexchange.com/ because other users mind find the answers more easily there.

closed 2024-10-25 by christophfroehlich