ros-controls/ros2_control #1811: [Question] When to use hardware_interface::HW_IF_EFFORT  and when to use hardware_interface::HW_IF_TORQUE?

firesurfer, 2024-10-25:
I have seen in all controllers / hardware interfaces that support some kind of torque control only the hardware_interface::HW_IF_EFFORT so far and thought - okey that is what we are using for torques. I now stumbled upon hardware_interface::HW_IF_TORQUE.

I am wondering now: What is the right  type to use for an actuator that supports reading/commanding torques? 
Is there some kind of rule/standard when to use what ?

bmagyar, 2024-10-25:
IMO if you have a torque sensor for feedback and can control torque of an
actuator, call it torque.

Effort is the more abstract term that can also mean current, acceleration,
torque or what have you. Also a non-linear combination of the above :D

saikishor, 2024-10-25:
Yes, it is to have meaningful interfaces in your components and controllers. We have an interface remapping feature in a PR: https://github.com/ros-controls/ros2_control/pull/1667 that can help you reuse the existing controllers with new interfaces by just remapping the interface.

firesurfer, 2024-11-21:
Thanks for the clarification !

closed 2024-11-21 by firesurfer

saikishor, 2024-11-21:
> Thanks for the clarification !

Thanks to you for bringing this up! :)