ros-controls/ros2_control #1818: Issues on connecting ros2_control to gazebo

Elarak, 2024-10-26:
Hello, I'm using ROS2 Jazzy and I have issues on connected ros2 to gazebo.
I want to control the 18 joints of my robot with forward_position_controller. 

I have 2 controllers initialized : ForwardCommandController and JointStateBroadcaster
ros2 control list_controllers
forward_position_controller forward_command_controller/ForwardCommandController  active
joint_state_broadcaster     joint_state_broadcaster/JointStateBroadcaster        active

We can see that all my joints are available
ros2 control list_hardware_interfaces
command interfaces
	joint_patte_1A_patte_1B/position [available] [claimed]
	joint_patte_1B_patte_1C/position [available] [claimed]
	joint_patte_1C_patte_1D/position [available] [claimed]
	joint_patte_2A_patte_2B/position [available] [claimed]
	joint_patte_2B_patte_2C/position [available] [claimed]
	joint_patte_2C_patte_2D/position [available] [claimed]
	joint_patte_3A_patte_3B/position [available] [claimed]
	joint_patte_3B_patte_3C/position [available] [claimed]
	joint_patte_3C_patte_3D/position [available] [claimed]
	joint_patte_4A_patte_4B/position [available] [claimed]
	joint_patte_4B_patte_4C/position [available] [claimed]
	joint_patte_4C_patte_4D/position [available] [claimed]
	joint_patte_5A_patte_5B/position [available] [claimed]
	joint_patte_5B_patte_5C/position [available] [claimed]
	joint_patte_5C_patte_5D/position [available] [claimed]
	joint_patte_6A_patte_6B/position [available] [claimed]
	joint_patte_6B_patte_6C/position [available] [claimed]
	joint_patte_6C_patte_6D/position [available] [claimed]

I can read the initial position of my joints with JointStateBroadcaster

And my publisher is publishing every 1 second between 2  valid positions.
We can see a typical message : 
ros2 topic echo forward_position_controller/commands
layout:
  dim: []
  data_offset: 0
data:
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 1.0
---

But on gazebo, the robot isn't moving. I don't have any changes on the joint_state topic. I don't have new messages on the terminal of my launch 
Am I missing something ?

christophfroehlich, 2024-10-26:
Is there any error output in the console log of the controller? The only check is the size of the data array must match.

Elarak, 2024-10-26:
I don't have any error for the forward: 
[INFO] [ros2-6]: process started with pid [4904]
[gazebo-1] [INFO] [1729979533.232722678] [controller_manager]: Loading controller : 'forward_position_controller' of type 'forward_command_controller/ForwardCommandController'
[gazebo-1] [INFO] [1729979533.232858422] [controller_manager]: Loading controller 'forward_position_controller'
[gazebo-1] [INFO] [1729979533.240767356] [controller_manager]: Controller 'forward_position_controller' node arguments: --ros-args --params-file ... --param use_sim_time:=true 
[ros2-6] Successfully loaded controller forward_position_controller
[gazebo-1] [INFO] [1729979533.338594584] [controller_manager]: Configuring controller: 'forward_position_controller'
[gazebo-1] [INFO] [1729979533.342060064] [forward_position_controller]: configure successful
[gazebo-1] [INFO] [1729979536.602491338] [controller_manager]: Activating controllers: [ forward_position_controller ]
[gazebo-1] [INFO] [1729979538.750409345] [forward_position_controller]: activate successful
[ros2-6] Successfully loaded controller forward_position_controller into state active

And for the broadcaster I have the same successfully loaded controller and I have warnings :
[gazebo-1] [WARN] [1729980256.923015232] [gz_ros_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[gazebo-1] [Dbg] [SDFFeatures.cc:332] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[gazebo-1] [Dbg] [SDFFeatures.cc:863] The geometry element of collision [joint] couldn't be created

 But after the initialization, I don't have any new logs when I'm using the publisher

christophfroehlich, 2024-10-27:
>  [gazebo-1] [Dbg] [SDFFeatures.cc:332] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes. [gazebo-1] [Dbg] [SDFFeatures.cc:863] The geometry element of collision [joint] couldn't be created

can you verify that your gazebo model works at all? I've never seen these warnings.

kdeshp-prof, 2025-01-02:
has there been any updates on this, I seem to get similär warning for my URDF and the controllers get active but the robot is not moving in simulation