ros-controls/ros2_control #1819: Cannot launch packet after update

vdovetzi, 2024-10-28:
Everything was fine, but today after today's Software Update on Ubuntu some of the gz packages were updated including gz_ros_control. And now after attempt of launching my moveit_gazebo_obb.launch.py it fails with the error:

```bash
[gazebo-1] [WARN] [1730100161.121211046] [gz_ros_control.resource_manager]: System hardware component 'panda_gazebo' from plugin 'gz_ros2_control/GazeboSimSystem' failed to initialize.
[gazebo-1] Stack trace (most recent call last) in thread 14870:
[gazebo-1] #12   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x734deb529c3b, in 
[gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x734deb49ca93, in 
[gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x734de60ecdb3, in 
[gazebo-1] #8    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x734da9318df6, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[gazebo-1] #7    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x734da9307bf9, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[gazebo-1] #6    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x734da93074ca, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[gazebo-1] #5    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x734dd4331da4, in 
[gazebo-1] #4    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x734dd42c117a, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[gazebo-1] #3    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x734dd42c6a7f, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[gazebo-1] #2    Object "/opt/ros/jazzy/lib/libhardware_interface.so", at 0x734dd41ed5a6, in hardware_interface::ResourceManager::get_components_status[abi:cxx11]()
[gazebo-1] #1    Object "/opt/ros/jazzy/lib/libhardware_interface.so", at 0x734dd421629a, in hardware_interface::System::get_name[abi:cxx11]() const
[gazebo-1] #0    Object "/home/vdovetzi/ws_moveit2/install/gz_ros2_control/lib/libgz_hardware_plugins.so", at 0x734dcd643663, in gz_ros2_control::GazeboSimSystem::initSim(std::shared_ptr<rclcpp::Node>&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned long, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned long> > >&, hardware_interface::HardwareInfo const&, gz::sim::v8::EntityComponentManager&, unsigned int)
[gazebo-1] Segmentation fault (Address not mapped to object [0xc9])
```

Today's updates:
```bash
2024-10-28 08:44:30 upgrade python3-gz-transport13:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:30 status half-configured python3-gz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked python3-gz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status half-installed python3-gz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked python3-gz-transport13:amd64 13.4.0-2~noble
2024-10-28 08:44:30 upgrade libgz-transport13-dev:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:30 status half-configured libgz-transport13-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status half-installed libgz-transport13-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-dev:amd64 13.4.0-2~noble
2024-10-28 08:44:30 upgrade libgz-transport13-log-dev:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:30 status half-configured libgz-transport13-log-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-log-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status half-installed libgz-transport13-log-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-log-dev:amd64 13.4.0-2~noble
2024-10-28 08:44:30 upgrade libgz-transport13-log:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:30 status half-configured libgz-transport13-log:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-log:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status half-installed libgz-transport13-log:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-log:amd64 13.4.0-2~noble
2024-10-28 08:44:30 upgrade libgz-transport13-parameters-dev:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:30 status half-configured libgz-transport13-parameters-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-parameters-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status half-installed libgz-transport13-parameters-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:30 status unpacked libgz-transport13-parameters-dev:amd64 13.4.0-2~noble
2024-10-28 08:44:31 upgrade libgz-transport13-parameters:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:31 status half-configured libgz-transport13-parameters:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13-parameters:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status half-installed libgz-transport13-parameters:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13-parameters:amd64 13.4.0-2~noble
2024-10-28 08:44:31 upgrade libgz-transport13-core-dev:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:31 status half-configured libgz-transport13-core-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13-core-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status half-installed libgz-transport13-core-dev:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13-core-dev:amd64 13.4.0-2~noble
2024-10-28 08:44:31 upgrade gz-transport13-cli:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:31 status half-configured gz-transport13-cli:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked gz-transport13-cli:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status half-installed gz-transport13-cli:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked gz-transport13-cli:amd64 13.4.0-2~noble
2024-10-28 08:44:31 upgrade libgz-transport13:amd64 13.4.0-1~noble 13.4.0-2~noble
2024-10-28 08:44:31 status half-configured libgz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status half-installed libgz-transport13:amd64 13.4.0-1~noble
2024-10-28 08:44:31 status unpacked libgz-transport13:amd64 13.4.0-2~noble
2024-10-28 08:44:37 upgrade ros-jazzy-gz-ros2-control:amd64 1.2.7-1noble.20240922.110347 1.2.7-1noble.20241007.230125
2024-10-28 08:44:37 status half-configured ros-jazzy-gz-ros2-control:amd64 1.2.7-1noble.20240922.110347
[... 73 lines cut ...]
2024-10-28 08:45:26 configure ros-jazzy-ros-gz-bridge:amd64 1.0.5-1noble.20241015.115607 <none>
2024-10-28 08:45:26 status unpacked ros-jazzy-ros-gz-bridge:amd64 1.0.5-1noble.20241015.115607
2024-10-28 08:45:26 status half-configured ros-jazzy-ros-gz-bridge:amd64 1.0.5-1noble.20241015.115607
2024-10-28 08:45:26 status installed ros-jazzy-ros-gz-bridge:amd64 1.0.5-1noble.20241015.115607
2024-10-28 08:45:26 configure libgz-transport13-dev:amd64 13.4.0-2~noble <none>
2024-10-28 08:45:26 status unpacked libgz-transport13-dev:amd64 13.4.0-2~noble
2024-10-28 08:45:26 status half-configured libgz-transport13-dev:amd64 13.4.0-2~noble
2024-10-28 08:45:26 status installed libgz-transport13-dev:amd64 13.4.0-2~noble
2024-10-28 08:45:27 configure ros-jazzy-ros-gz-image:amd64 1.0.5-1noble.20241015.121738 <none>
2024-10-28 08:45:27 status unpacked ros-jazzy-ros-gz-image:amd64 1.0.5-1noble.20241015.121738
2024-10-28 08:45:27 status half-configured ros-jazzy-ros-gz-image:amd64 1.0.5-1noble.20241015.121738
2024-10-28 08:45:27 status installed ros-jazzy-ros-gz-image:amd64 1.0.5-1noble.20241015.121738
2024-10-28 08:45:27 configure ros-jazzy-ros-gz-sim-demos:amd64 1.0.5-1noble.20241015.121936 <none>
2024-10-28 08:45:27 status unpacked ros-jazzy-ros-gz-sim-demos:amd64 1.0.5-1noble.20241015.121936
2024-10-28 08:45:27 status half-configured ros-jazzy-ros-gz-sim-demos:amd64 1.0.5-1noble.20241015.121936
2024-10-28 08:45:27 status installed ros-jazzy-ros-gz-sim-demos:amd64 1.0.5-1noble.20241015.121936
[... cut ...]
2024-10-28 08:45:27 status half-configured ros-jazzy-gz-ros2-control:amd64 1.2.7-1noble.20241007.230125
2024-10-28 08:45:27 status installed ros-jazzy-gz-ros2-control:amd64 1.2.7-1noble.20241007.230125
2024-10-28 08:45:27 configure ros-jazzy-ros-gz:amd64 1.0.5-1noble.20241015.122451 <none>
2024-10-28 08:45:27 status unpacked ros-jazzy-ros-gz:amd64 1.0.5-1noble.20241015.122451
2024-10-28 08:45:27 status half-configured ros-jazzy-ros-gz:amd64 1.0.5-1noble.20241015.122451
2024-10-28 08:45:27 status installed ros-jazzy-ros-gz:amd64 1.0.5-1noble.20241015.122451
```

vdovetzi, 2024-10-29:
@saikishor 

I will check it later. Building from source did a trick for me)

saikishor, 2024-10-29:
> I will check it later. Building from source did a trick for me)

Thanks for letting us know. If this is the case, the debian version should also work. I'll proceed to close this issue. If you continue to have the issue, please write back here so we can open it again

closed 2024-10-29 by saikishor