ros-controls/ros2_control #1821: Add ros2_control_node to documentation

christophfroehlich, 2024-10-29:
I realized that we don't mention the ros2_control_node not even once in the docs. Should we?

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_control/pull/1820#pullrequestreview-2402999262_

fmauch, 2024-10-30:
This is not entirely true, as we mention it e.g. [here](https://control.ros.org/rolling/doc/getting_started/getting_started.html#controller-manager) but it could definitely placed more visible.

Should we add and explain a minimal launch file with 
 - description
 - controller config file
 - robot state publisher
 - ros2_control_node
 - controller spawner
 - RViz

This almost boils down to the launch setup from example1, so we could just try to explain that a bit more in detail and link to that explanation from the getting_started page?

christophfroehlich, 2024-10-30:
ah, you are right. In the demos we don't explain what this node is for. I would maybe extend the get started section a bit, and describe parameters like introduced with #1820? And then link to example_1.

saikishor, 2024-10-30:
> ah, you are right. In the demos we don't explain what this node is for. I would maybe extend the get started section a bit, and describe parameters like introduced with #1820? And then link to example_1.

Yes! some overview of the parameters and what is is used for and a link to an example should do

saikishor, 2024-12-15:
Closed with #1890

closed 2024-12-15 by saikishor