ros-controls/ros2_control #1880: How to build ros2_control master from source for humble on 22.04

lcbw, 2024-11-19:
Distro: Humble | OS: Ubuntu 22.04 | Build: Source | Branch: master 

What I'm trying to do:
I'm trying to build ros2_control and its dependencies from source to obtain the export_state_interfaces functionality as a part of the ChainedControllerInterface type. 

Setup:
I've vcs import'd all of the repositories specified in the ros2_controllers.humble.repos on ros2_controllers origin/master and ran rosdep install on all of the repositories therein, except for ros2_control, which I've imported on origin/master, and still my build fails. 

I'm using the build command colcon build --cmake-args -DBUILD_TESTING=OFF

Any advisement or clarification would be appreciated - thank you!

christophfroehlich, 2024-11-19:
What is the error you are getting?

You have to build the whole stack, the instructions are written [here](https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source) under "Development Version".

lcbw, 2024-11-19:
I followed the instructions you linked @christophfroehlich and I still get this same error: 
``` 
[ 66%] Building CXX object CMakeFiles/joint_state_broadcaster.dir/src/joint_state_broadcaster.cpp.o
In file included from /home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:21,
                 from /home/ros-industrial/workspaces/ros2_control/src/ros-controls/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:24,
                 from /home/ros-industrial/workspaces/ros2_control/src/ros-controls/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:268:61: error: ‘SharedPtr’ is not a member of ‘hardware_interface::CommandInterface’
  268 |   virtual std::vector<hardware_interface::CommandInterface::SharedPtr>
      |                                                             ^~~~~~~~~
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:268:70: error: template argument 1 is invalid
  268 |   virtual std::vector<hardware_interface::CommandInterface::SharedPtr>
      |                                                                      ^
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:268:70: error: template argument 2 is invalid
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:278:59: error: ‘ConstSharedPtr’ is not a member of ‘hardware_interface::StateInterface’
  278 |   virtual std::vector<hardware_interface::StateInterface::ConstSharedPtr>
      |                                                           ^~~~~~~~~~~~~~
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:278:73: error: template argument 1 is invalid
  278 |   virtual std::vector<hardware_interface::StateInterface::ConstSharedPtr>
      |                                                                         ^
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface_base.hpp:278:73: error: template argument 2 is invalid
In file included from /home/ros-industrial/workspaces/ros2_control/src/ros-controls/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:24,
                 from /home/ros-industrial/workspaces/ros2_control/src/ros-controls/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:50:51: error: ‘ConstSharedPtr’ is not a member of ‘hardware_interface::StateInterface’
   50 |   std::vector<hardware_interface::StateInterface::ConstSharedPtr> export_state_interfaces() final;
      |                                                   ^~~~~~~~~~~~~~
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:50:65: error: template argument 1 is invalid
   50 |   std::vector<hardware_interface::StateInterface::ConstSharedPtr> export_state_interfaces() final;
      |                                                                 ^
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:50:65: error: template argument 2 is invalid
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:58:53: error: ‘SharedPtr’ is not a member of ‘hardware_interface::CommandInterface’
   58 |   std::vector<hardware_interface::CommandInterface::SharedPtr> export_reference_interfaces() final;
      |                                                     ^~~~~~~~~
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:58:62: error: template argument 1 is invalid
   58 |   std::vector<hardware_interface::CommandInterface::SharedPtr> export_reference_interfaces() final;
      |                                                              ^
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:58:62: error: template argument 2 is invalid
gmake[2]: *** [CMakeFiles/joint_state_broadcaster.dir/build.make:76: CMakeFiles/joint_state_broadcaster.dir/src/joint_state_broadcaster.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/joint_state_broadcaster.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
--- stderr: joint_state_broadcaster
In file included from /home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:21,
[... cut ...]
/home/ros-industrial/workspaces/ros2_control/install/controller_interface/include/controller_interface/controller_interface/controller_interface.hpp:58:62: error: template argument 2 is invalid
gmake[2]: *** [CMakeFiles/joint_state_broadcaster.dir/build.make:76: CMakeFiles/joint_state_broadcaster.dir/src/joint_state_broadcaster.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/joint_state_broadcaster.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< joint_state_broadcaster [11.7s, exited with code 2]

```

christophfroehlich, 2024-11-19:
It seems that something does not work with the overlay of the classes. the SharedPtr is defined in hardware_interface/handle.hpp
In case you haven't tried, delete build and install folder first.
If this doesn't help, uninstall the binary humble packages of ros2_control before you run colcon build. (they might have been installed with rosdep if you haven't added the --ignore-src argument.

lcbw, 2024-11-19:
I think the issue was some other conflicting libgazebo versions - closing this. thank you for your help @christophfroehlich