ros-controls/ros2_control #1882: Cleanup all the deprecations warning from the recent ros2_control variant changes

saikishor, 2024-11-20:
Recently after the new variants feature in Handles (https://github.com/ros-controls/ros2_control/pull/1688) We have a bunch of deprecation warnings that needs to be cleanup. This could be a nice first issue.

MartinPeris, 2024-11-21:
I'd like to take this one. Quick question. For example in this situation:

```
/root/ws_ros2_control/src/ros2_control/hardware_interface/include/hardware_interface/system_interface.hpp: In member function ‘double hardware_interface::SystemInterface::get_state(const std::string&) const’:
/root/ws_ros2_control/src/ros2_control/hardware_interface/include/hardware_interface/system_interface.hpp:410:56: warning: ‘double hardware_interface::Handle::get_value() const’ is deprecated: Use bool get_value(double & value) instead to retrieve the value. [-Wdeprecated-declarations]
  410 |     return system_states_.at(interface_name)->get_value();
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
```

The original code is:

```c++
double get_state(const std::string & interface_name) const
  {
    return system_states_.at(interface_name)->get_value();
  }
```

Since the new API signature is `bool get_value(double &value)`, how do you want me to handle the `bool` returned by `get_value`?

Ignoring it would not be an option for me, so this leaves 3 possibilities:

1. Throw an Exception
2. Return a Sentinel Value (e.g., NaN)
3. Log an Error and Return Default Value

Either 1 or 2 would be acceptable for me (leaning towards 2, bit that is just personal preference). In any case, new unit tests will have to be added... I charge extra karma for that :P

MartinPeris, 2024-11-21:
mmmm looking at the ~~documentation~~ source code looks like a quiet NaN is the way to go:

```c++
[[deprecated("Use bool get_value(double & value) instead to retrieve the value.")]]
  double get_value() const
  {
    std::shared_lock<std::shared_mutex> lock(handle_mutex_, std::try_to_lock);
    if (!lock.owns_lock())
    {
      return std::numeric_limits<double>::quiet_NaN();
    }
    // BEGIN (Handle export change): for backward compatibility
    // TODO(Manuel) return value_ if old functionality is removed
    THROW_ON_NULLPTR(value_ptr_);
    return *value_ptr_;
    // END
  }
```

saikishor, 2024-12-22:
Hello @MartinPeris!

First of all, I want to apologize for not getting back to your earlier.
 
For your question! I think we should also deprecate them that have void and instead they have to return bool the one from the handles. and so, we will have it similar to the Handles. Something like the following:

https://github.com/ros-controls/ros2_control/blob/038a05f512a58e9b23f9d282395c992fde3e6c36/hardware_interface/include/hardware_interface/handle.hpp#L175-L188

So, the user will be able to know if what he gets is valid or not. 

You can add the [gcc ignore deprecated warnings](https://stackoverflow.com/a/13459924) to the deprecated methods, so we don't see them any more.

What do you think of this approach? @MartinPeris @ros-controls/ros2-maintainers

AdityaPawar162, 2025-03-05:
Hi, I’m trying to work on this issue and replicate it. However, when I build ros2_control from source using colcon build on Humble, I don't see any warnings appearing.

AdityaPawar162, 2025-03-05:
Thanks for the update! I see that the warnings in transmissions are being addressed in #1845. Is there any other good first issue available that I can work on?

saikishor, 2025-03-05:
Hello @AdityaPawar162 

Maybe for now, try to get something from the [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos/issues) repository. We have a couple over there. In the meantime, If we find something else, we will open an issue with the tag again.

saikishor, 2025-08-19:
> @saikishor is this done?

@christophfroehlich it is not done.

christophfroehlich, 2025-09-26:
@saikishor we have lots of deprecation warnings in the jazzy build. Either, we remove the deprecation warnings or fix
https://github.com/ros-controls/ros2_control/pull/2583
https://github.com/ros-controls/ros2_controllers/pull/1934

saikishor, 2025-09-27:
@christophfroehlich I've a feeling that it is already handled in ros2_control repo, we just have to check and merge the ros2_controllers PR