ros-controls/ros2_control #1883: [ Jazzy ] - Startup failes sometimes with: "Importing state interfaces for the hardware '...' returned no state interfaces."

firesurfer, 2024-11-21:
**Describe the bug**

Startup of hardware interface fails sometimes with:

> [ros2_control_node-19] [WARN] [1732175145.582300560] [controller_manager.resource_manager]: Importing state interfaces for the hardware 'ur_top' returned no state interfaces.

This message is printed from all hardware interfaces. 

I can then go ahead restart everything and it works just fine. This appeared after to update to jazzy.
Afterwards the controller_manager node exits. 

**To Reproduce**

Have a system with multiple hardware interfaces. At some point this issue will occur. 

**Expected behavior**

Startup should always work just fine ;) 

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04 - Docker container based on upstream ros:jazzy image
 - Version Jazzy
- Quite a lot of hardware interfaces and controllers

**Additional context**

Happens with or without my custom adaptions to the admittance_controller and the JTC.

firesurfer, 2024-11-25:
I just updated my jazzy installation (for the controller_manager this would be 4.20.0 and JTC 4.16.0 as a reference).

I get this warning now continuously for all hardware interfaces at every start of the software stack.
In addition loading the JTC fails with:

```
[ros2_control_node-19] [ERROR] [1732534783.652359103] [joint_trajectory_controller_top]: Caught exception in callback for transition 10
[ros2_control_node-19] [ERROR] [1732534783.652476585] [joint_trajectory_controller_top]: Original error: Resource temporarily unavailable
[ros2_control_node-19] [WARN] [1732534783.652808481] [joint_trajectory_controller_top]: Error occurred while doing error handling.
[ros2_control_node-19] [ERROR] [1732534783.654384242] [controller_manager]: After configuring, controller 'joint_trajectory_controller_top' is in state 'unconfigured' , expected inactive.

```

I am not sure if these errors are related. But with the latest update ros2_control is basically broken for our setup

christophfroehlich, 2024-11-25:
just to clarify: you have now the latest jazzy binary packages, not built them from source?

firesurfer, 2024-11-25:
Yes these are the latest jazzy binary packages.

I just did some additional tests. The issue with the JTC seems not to be related with the initial issue. Other controllers seem to work fine.  I will create a separate issue for it in the ros2_controllers repository.

firesurfer, 2024-11-28:
I close this issue as I am pretty sure what I actually saw there is what I described in #1891 
It was just by chance that these warnings were catching my eye then.

closed 2024-11-28 by firesurfer