ros-controls/ros2_control #1889: `realtime_tools/realtime_helpers.hpp` referenced but not recognized

aalmrad, 2024-11-26:
**Describe the bug**
The `realtime_tools/realtime_helpers.hpp` header file is not recognized in the `controller_manager/src/ros2_control_node.cpp` file.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the humble branch of the repository to the source directory of a ros2 workspace
2. build the workspace

**Expected behavior**
Clean and smooth build process with no errors

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
- ROS2: Humble

saikishor, 2024-11-26:
@aalmrad Can you please let us know the realtime_tools version you are using?

yuvansharma, 2024-11-26:
I am also getting this error. I have version 2.7.0 installed for ros-humble-realtime-tools.

christophfroehlich, 2024-11-26:
Please update realtime_tools. Either by compiling from source (maybe you have to uninstall the binary first) or update the released version (which is 2.8.1).
https://index.ros.org/p/realtime_tools/github-ros-controls-realtime_tools/#humble

closed 2024-11-26 by christophfroehlich