ros-controls/ros2_control #1891: hardware_interface - read/write are called before on_activate has finished

firesurfer, 2024-11-28:
**Describe the bug**

During our migration from Iron to Jazzy I ran into the issue of having random segfaults during startup in one of our hardware interfaces. 

The cause of the issue seems to be that the read/write methods are called before the on_activate method has finished (It is actually random what is executed first). The segfault in our case was a pointer being accessed in the the read method that is initialized in the on_activate method

**To Reproduce**

Write a hardware interface, and simply check if read/write or on_activate is called first. 

**Expected behavior**

`on_activate` should be called and should have finished before the read/write methods of a hardware interface are called. 

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04 - ROS2 installation in container
 - Version: Jazzy binary install, hardware_interface: 4.20.0

christophfroehlich, 2024-11-28:
I think it's not a bug, it's a feature.

AFAIK, the idea was that joint values are sent only in activated mode but all non-joint interfaces are read/written from configured on. But I think this distinction is not implemented yet and always sent.

saikishor, 2025-04-02:
There is a small fix in this PR : https://github.com/ros-controls/ros2_control/pull/2046, that should avoid the read/write subsequent calls at the startup. For this reason, I'll go ahead and close this. This is also tested on our hardware :)