ros-controls/ros2_control #1906: Statically allocate string concatinations using FMT

saikishor, 2024-12-02:
could we please add a follow-up issue to make this statically allocated by using FMT?

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_control/pull/1871#discussion_r1866454408_

mini-1235, 2025-04-15:
I am not sure if I understand this issue correctly, does it mean to add something like `fmt::format` to replace the original `const std::string`?

christophfroehlich, 2025-04-19:
I think so, yes @saikishor?

saikishor, 2025-04-20:
@mini-1235 Yes, we need to do it similar way, the only thing is it is missing `FMT_COMPILE` . You can check more details here : https://fmt.dev/9.1.0/api.html#format-string-compilation

mini-1235, 2025-04-26:
> [@mini-1235](https://github.com/mini-1235) Yes, we need to do it similar way, the only thing is it is missing `FMT_COMPILE` . You can check more details here : https://fmt.dev/9.1.0/api.html#format-string-compilation

@saikishor thanks for the guidance! I have opened a PR for it

PR #2249 (2025-05-19): Statically allocate string concatenations using FMT formatting (backport #2205)
closes #1906<hr>This is an automatic backport of pull request #2205 done by [Mergify](https://mergify.com).