ros-controls/ros2_control #1921: Not acceptable command interfaces combination ros2_control

JagaMuthu, 2024-12-07:
I having trouble with the ro2 control issue. It repeat the same error. 
[gzserver-1] [INFO] [1733571413.941938155] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[gzserver-1] [ERROR] [1733571413.961346237] [resource_manager]: Not acceptable command interfaces combination:

[image]

      
    <gazebo>
        <plugin filename="libgazebo_ros2_control.so" name="gazebo_ros2_control">
            <robot_param>robot_description</robot_param>
            <robot_param_node>robot_state_publisher</robot_param_node>
            <parameters>$(find exo_model)/config/robot_control.yaml</parameters>
        </plugin>
    </gazebo>

    <ros2_control name="RobotSystem" type="system">
        <hardware>
            <plugin>gazebo_ros2_control/GazeboSystem</plugin>
        </hardware>
        <joint name="Joint_2">
            <command_interface name="position">
            <param name="min">-3.14</param>
            <param name="max">3.14</param>
            </command_interface>
            <state_interface name="position">
            <param name="initial_value">0</param>
            </state_interface>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>
        </joint>
         <joint name="Joint_3">
            <command_interface name="position">
            <param name="min">-3.14</param>
            <param name="max">3.14</param>
            </command_interface>
            <state_interface name="position">
            <param name="initial_value">0</param>
            </state_interface>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>
        </joint>
         <joint name="Joint_4">
            <command_interface name="position">
            <param name="min">-3.14</param>
            <param name="max">3.14</param>
            </command_interface>
            <state_interface name="position">
            <param name="initial_value">0</param>
            </state_interface>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>
        </joint>
         <joint name="Joint_5">
            <command_interface name="position">
            <param name="min">-3.14</param>
            <param name="max">3.14</param>
            </command_interface>
            <state_interface name="position">
            <param name="initial_value">0</param>
            </state_interface>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>
        </joint>
    </ros2_control>
</robot>

 My launch file
#!/usr/bin/env python3
import os
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription, DeclareLaunchArgument, SetEnvironmentVariable, ExecuteProcess
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node
from launch.substitutions import Command, LaunchConfiguration
from launch_ros.parameter_descriptions import ParameterValue
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():
    # Set GAZEBO_MODEL_PATH for custom models
    gazebo_model_path = os.path.join(
        get_package_share_directory("exo_model"), "models"
    )
    gazebo_env = SetEnvironmentVariable(
        "GAZEBO_MODEL_PATH", gazebo_model_path
    )

    # Path to URDF file
    urdf_file = os.path.join(
        get_package_share_directory("exo_model"), "urdf", "exo_model_control.urdf"
    )
    if not os.path.exists(urdf_file):
        raise FileNotFoundError(f"URDF file not found: {urdf_file}")
    print(f"Using URDF file: {urdf_file}")

    # Robot description parameter
    robot_description = ParameterValue(
        Command(['xacro ', urdf_file]), value_type=str
    )

    # Gazebo launch description (ensures Gazebo 11 is launched)
    gazebo = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([os.path.join(
            get_package_share_directory('gazebo_ros'),
            'launch',
            'gazebo.launch.py'
        )]),
        launch_arguments={'verbose': 'true'}.items()
    )

    # Robot State Publisher Node
    robot_state_publisher_node = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        output='screen',
        parameters=[{
            'robot_description': robot_description,
            'use_sim_time': LaunchConfiguration('use_sim_time')
        }]
    )

    # Spawn Entity Node to load the robot into Gazebo
    spawn_entity_node = Node(
        package='gazebo_ros',
        executable='spawn_entity.py',
        name='urdf_spawner',
        output='screen',
        arguments=[
            '-topic', '/robot_description',  # Subscribe to the /robot_description topic
            '-entity', 'exo_model',
        ]
    )

    # Load joint state broadcaster
    load_joint_state_broadcaster = ExecuteProcess(
        cmd=[
            'ros2', 'control', 'load_controller',
            '--set-state', 'active',
            'joint_state_broadcaster'
        ],
        output='screen'
    )

    # Load joint trajectory controller
    load_joint_trajectory_controller = ExecuteProcess(
        cmd=[
            'ros2', 'control', 'load_controller',
            '--set-state', 'active',
            'joint_trajectory_controller'
        ],
        output='screen'
    )

    return LaunchDescription([
        DeclareLaunchArgument(
            'use_sim_time',
            default_value='true',
            description='Use simulation clock if true'
        ),
        gazebo_env,  # Set GAZEBO_MODEL_PATH
        gazebo,  # Launch Gazebo 11
        robot_state_publisher_node,  # Publish robot's state
        spawn_entity_node,  # Spawn the robot in Gazebo
        load_joint_state_broadcaster,  # Load joint state broadcaster
        load_joint_trajectory_controller,  # Load joint trajectory controller
    ])
 
The yaml file
controller_manager
  ros__parameters:
        update_rate: 100
            
      joint_trajectory_controller:
        type: joint_trajectory_controller/JointTrajectoryController
  
      joint_state_broadcaster:
         type: joint_state_broadcaster/JointStateBroadcaster
 
 joint_trajectory_controller:
     ros__parameters: 
        joints:
             - Joint_2
             - Joint_3
             - Joint_4
             - Joint_5
     interface_name: position
     command_interfaces:
             - position
     state_interfaces:
             - position
             -velocity

saikishor, 2024-12-10:
Hello!

Before launching any controller, can you check the available hardware interfaces with `ros2 control list_hardware_interfaces` ?

Thank you!

qleonardolp, 2025-03-19:
@JagaMuthu consider remove the parameter `interface_name` on joint_trajectory_controller. The params `command_interfaces` and `state_interfaces` are enough.

christophfroehlich, 2025-03-19:
@JagaMuthu please don't post screenshot of text next time, and take a bit of time for properly format your question (did that for you).
As @qleonardolp answered, your error likely comes form the JTC configuration.

closed 2025-03-19 by christophfroehlich