ros-controls/ros2_control #1926: `rw_rate` is not accessible from the HardwareComponent

Noel215, 2024-12-10:
Hi,

I've been trying to obtain the `rw_rate` of my HardwareComponent by calling the [get_hardware_info()](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/system_interface.hpp#L439) method, but when I get the value of the `rw_rate`, it is always zero.

saikishor, 2024-12-10:
Thank you for reporting it

xibeisiber, 2025-01-06:
there is rw_rate in hardwareinfo struct on master branch. Will this be merged to humble branch?

christophfroehlich, 2025-01-06:
no we don't have plans to backport this to humble. but you can compile the rolling development version from source on humble

xibeisiber, 2025-01-06:
> no we don't have plans to backport this to humble. but you can compile the rolling development version from source on humble

ok. thanks

saikishor, 2025-01-06:
@xibeisiber you can find more information on the page : https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source under "Development version".

Thank you