ros-controls/ros2_control #1936: `test_components_with_different_read_write_freq_on_deactivate` is flaky on RHEL

christophfroehlich, 2024-12-12:
https://github.com/ros-controls/ros2_control/actions/runs/12291365698/job/34300007474?pr=1935
```
    [ RUN      ] ResourceManagerTestReadWriteDifferentReadWriteRate.test_components_with_different_read_write_freq_on_deactivate
    [INFO] [1733986838.013298496] [ResourceManagerTest.resource_manager]: Loading hardware 'TestActuatorHardware' 
    [INFO] [1733986838.013526963] [ResourceManagerTest.resource_manager]: Loaded hardware 'TestActuatorHardware' from plugin 'test_actuator'
    [INFO] [1733986838.013574341] [ResourceManagerTest.resource_manager]: Initialize hardware 'TestActuatorHardware' 
    [INFO] [1733986838.013624846] [ResourceManagerTest.resource_manager]: Successful initialization of hardware 'TestActuatorHardware'
    [INFO] [1733986838.013698143] [ResourceManagerTest.resource_manager]: Loading hardware 'TestSensorHardware' 
    [INFO] [1733986838.013864013] [ResourceManagerTest.resource_manager]: Loaded hardware 'TestSensorHardware' from plugin 'test_sensor'
    [INFO] [1733986838.013901833] [ResourceManagerTest.resource_manager]: Initialize hardware 'TestSensorHardware' 
    [INFO] [1733986838.013938502] [ResourceManagerTest.resource_manager]: Successful initialization of hardware 'TestSensorHardware'
    [INFO] [1733986838.013967706] [ResourceManagerTest.resource_manager]: Loading hardware 'TestSystemHardware' 
    [INFO] [1733986838.014130661] [ResourceManagerTest.resource_manager]: Loaded hardware 'TestSystemHardware' from plugin 'test_system'
    [INFO] [1733986838.014162871] [ResourceManagerTest.resource_manager]: Initialize hardware 'TestSystemHardware' 
    [INFO] [1733986838.014224987] [ResourceManagerTest.resource_manager]: Successful initialization of hardware 'TestSystemHardware'
    [INFO] [1733986838.014367874] [resource_manager]: 'configure' hardware 'TestActuatorHardware' 
    [INFO] [1733986838.014380768] [resource_manager]: Successful 'configure' of hardware 'TestActuatorHardware'
    [INFO] [1733986838.014392390] [resource_manager]: 'activate' hardware 'TestActuatorHardware' 
    [INFO] [1733986838.014398932] [resource_manager]: Successful 'activate' of hardware 'TestActuatorHardware'
    [INFO] [1733986838.014413449] [resource_manager]: 'configure' hardware 'TestSensorHardware' 
    [INFO] [1733986838.014420433] [resource_manager]: Successful 'configure' of hardware 'TestSensorHardware'
    [INFO] [1733986838.014427876] [resource_manager]: 'activate' hardware 'TestSensorHardware' 
    [INFO] [1733986838.014433617] [resource_manager]: Successful 'activate' of hardware 'TestSensorHardware'
    [INFO] [1733986838.014446271] [resource_manager]: 'configure' hardware 'TestSystemHardware' 
    [INFO] [1733986838.014453074] [resource_manager]: Successful 'configure' of hardware 'TestSystemHardware'
    [INFO] [1733986838.014465687] [resource_manager]: 'activate' hardware 'TestSystemHardware' 
    [INFO] [1733986838.014472319] [resource_manager]: Successful 'activate' of hardware 'TestSystemHardware'
    [INFO] [1733986838.014557679] [resource_manager]: 'deactivate' hardware 'TestSystemHardware' 
    [INFO] [1733986838.014572136] [resource_manager]: Successful 'deactivate' of hardware 'TestSystemHardware'
    [INFO] [1733986838.014580822] [resource_manager]: 'deactivate' hardware 'TestActuatorHardware' 
    [INFO] [1733986838.014586954] [resource_manager]: Successful 'deactivate' of hardware 'TestActuatorHardware'
    [INFO] [1733986838.014595359] [resource_manager]: 'deactivate' hardware 'TestSensorHardware' 
    [INFO] [1733986838.014601731] [resource_manager]: Successful 'deactivate' of hardware 'TestSensorHardware'
    /__w/ros2_control/ros2_control/src/ros-controls/ros2_control/hardware_interface_testing/test/test_resource_manager.cpp:1847: Failure
    Expected equality of these values:
      state_itfs[1].get_value()
        Which is: 10
      prev_system_state_value
        Which is: 0
    
    [  FAILED  ] ResourceManagerTestReadWriteDifferentReadWriteRate.test_components_with_different_read_write_freq_on_deactivate (59 ms)
```

saikishor, 2024-12-12:
Thank you for the issue.

christophfroehlich, 2024-12-19:
https://github.com/ros-controls/ros2_control/actions/runs/12410415493/attempts/1?pr=1941

closed 2025-07-01 by github-actions[bot]