ros-controls/ros2_control #1956: Designing a System that Exposes Multiple CAN Devices to ROS while Interfacing with Master Device via Serial in Hardware

kyleoptimotive, 2024-12-17:
**Is your feature request related to a problem? Please describe.**
I'm looking for input towards the best practices for accomplishing what I've described above. I've confirmed that I can 1. expose _one_ device to ROS and 2. Talk to every device that is connected to the first device via CAN bus. Now I am seeking to expose all devices to ROS while being limited to one serial connection.

**Describe the solution you'd like**
On a high level, I would like each of the devices exposed to ROS so that they can be run by different controllers without needing to create new code/systems for each one (for example Ackermann, velocity/position, differential).

Hardware Configuration:
`(host system) <---serial---> (device_0) <---CAN---> (device_1, ..., device_n)`

Desired Behaviour in ROS:
`(host system) <-> (device_0, device_1, ... device_n)`

**Describe alternatives you've considered**
These are the ideas/options I have so far and I would appreciate insight.

1. Create virtual hardware interfaces through a Node that has the ability to communicate to all CAN devices.
* This node would publish/subscribe to topics that correspond to the hardware devices and would manage the mapping between those topics and communicating over serial/CAN.
* For example: Lets say we want to set the velocity of a device with CAN ID 1 `dev_1` and we are serially connected to device with CAN ID 0 `dev_0`. ROS would see topics like `/dev_0/velocity/command` and `/dev_1/velocity/command` created by the intermediate node and publish to the topic pertaining to `dev_1`. Then the intermediate node would see that message and pass it through `dev_0` to eventually reach `dev_1` over CAN.

2. Create a custom ROS control system hardware for each robot using these CAN-chained controllers.

3. Any other suggestions you may have.

**Additional context**
I am using ROS Humble and Ubuntu 22.04.

My specific use case of this is I am trying to control multiple VESC motor controllers using ROS2 Control. The VESC controllers are connected to each other over a CAN bus and the master VESC is connected to the system running ROS over USB serial.

I am using [sbgisen/vesc](https://github.com/sbgisen/vesc/tree/humble-devel) to create the interface between my VESC controller and ROS.

saikishor, 2024-12-18:
Hello!

Why don't you use a System Hardware Component interface for this approach?. This can help you handle all your multi-actuator communication in a single plugin.

bmagyar, 2024-12-19:
I think this is a topic we've been flirting with since 2020. "Comms". 
I think the OP's main issue is that it'd be good to transparently expose those devices without having to know where they come from, it's all the same bus after all. There may be more, there may be less devices on it.

The ideal architecture would have a `CANCommsDevice` configured in the ros2_control tag which is then shared between multiple system or actuator components to do their communication. Is this what you are looking for @kyleoptimotive ?

kyleoptimotive, 2024-12-19:
Thank you for the responses. I think the System Hardware Component interface might work for what I need, I'll try it out.

On @bmagyar 's point, is this something that exists? It sounds like a good idea, could you highlight the differences between this "Comms" concept and what a System Hardware Component offers?

firesurfer, 2025-02-07:
> The ideal architecture would have a CANCommsDevice configured in the ros2_control tag which is then shared between multiple system or actuator components to do their communication.

I am actually looking for the same but for ethercat. 

If I read @bmagyar answer correctly this is currently not possible?
My fallback currently is to create a singleton for the ethercat master in our ethercat library and use that from multiple hardware interfaces. But this is far from ideal. Would be nicer if it would be possible to share resources between hardware interfaces.

qleonardolp, 2025-03-17:
> I am actually looking for the same but for ethercat.
>
> If I read [@bmagyar](https://github.com/bmagyar) answer correctly this is currently not possible? My fallback currently is to create a singleton [...]

Consider this package https://github.com/ICube-Robotics/ethercat_driver_ros2.
There is some examples on how to use this packge [here](https://github.com/ICube-Robotics/ethercat_driver_ros2_examples)