ros-controls/ros2_control #1957: check_the_controller_update_rate is flaky

christophfroehlich, 2024-12-18:
I saw that several times now, mostly on humble 

https://github.com/ros-controls/ros2_control_ci/actions/runs/12385126242/job/34570846255

```
     1 FAILED TEST
    -- run_test.py: return code 1
    -- run_test.py: inject classname prefix into gtest result file '/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/build/controller_manager/test_results/controller_manager/test_controller_manager.gtest.xml'
    -- run_test.py: verify result file '/home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/build/controller_manager/test_results/controller_manager/test_controller_manager.gtest.xml'
  >>>
build/controller_manager/test_results/controller_manager/test_controller_manager.gtest.xml: 30 tests, 0 errors, 1 failure, 0 skipped
- controller_manager.per_controller_update_rate_check/TestControllerUpdateRates check_the_controller_update_rate/10
  <<< failure message
    /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/src/ros-controls/ros2_control/controller_manager/test/test_controller_manager.cpp:736
    Value of: test_controller->update_period_.seconds()
    Expected: (is > 0.011) and (is < 0.0216667)
      Actual: 0.022103 (of type double)
    update_counter: 117 desired controller period: 0.011111111111111112 actual controller period: 0.022103009999999999
    /home/runner/work/ros2_control_ci/ros2_control_ci/.work/target_ws/src/ros-controls/ros2_control/controller_manager/test/test_controller_manager.cpp:736
    Value of: test_controller->update_period_.seconds()
    Expected: (is > 0.011) and (is < 0.0216667)
      Actual: 0.022103 (of type double)
    update_counter: 118 desired controller period: 0.011111111111111112 actual controller period: 0.022103009999999999
  >>>
```

christophfroehlich, 2025-01-03:
https://github.com/ros-controls/ros2_control_ci/issues/217

christophfroehlich, 2025-04-01:
https://github.com/ros-controls/ros2_control_ci/issues/290

christophfroehlich, 2025-04-22:
https://github.com/ros-controls/ros2_control_ci/issues/315

christophfroehlich, 2025-05-17:
https://github.com/ros-controls/ros2_control/issues/2225

christophfroehlich, 2025-05-28:
https://github.com/ros-controls/ros2_control_ci/issues/344

christophfroehlich, 2025-05-30:
https://github.com/ros-controls/ros2_control/issues/2304

christophfroehlich, 2025-06-26:
https://github.com/ros-controls/ros2_control_ci/issues/383

christophfroehlich, 2025-08-06:
https://github.com/ros-controls/ros2_control/issues/2442

christophfroehlich, 2025-09-17:
https://github.com/ros-controls/ros2_control_ci/issues/462

christophfroehlich, 2025-09-18:
https://github.com/ros-controls/ros2_control/issues/2549

christophfroehlich, 2025-09-19:
https://github.com/ros-controls/ros2_control/issues/2568

saikishor, 2025-09-19:
@christophfroehlich I've a feeling that https://github.com/ros-controls/ros2_control/pull/2557 might fix these tests too

closed 2025-09-22 by saikishor

christophfroehlich, 2025-09-24:
We also need to backport this stuff to humble
https://github.com/ros-controls/ros2_control_ci/issues/471

christophfroehlich, 2025-09-25:
This is rolling stack on humble:
https://github.com/ros-controls/ros2_control_ci/issues/472

christophfroehlich, 2025-09-25:
https://github.com/ros-controls/ros2_control/issues/2581

christophfroehlich, 2025-09-25:
https://github.com/ros-controls/ros2_control_ci/issues/473

christophfroehlich, 2025-09-28:
https://github.com/ros-controls/ros2_control_ci/issues/480

christophfroehlich, 2025-10-01:
https://github.com/ros-controls/ros2_control_ci/issues/488

christophfroehlich, 2025-10-01:
https://github.com/ros-controls/ros2_control/issues/2595

christophfroehlich, 2025-10-03:
https://github.com/ros-controls/ros2_control/issues/2607
https://github.com/ros-controls/ros2_control_ci/issues/492
https://github.com/ros-controls/ros2_control_ci/issues/493
https://github.com/ros-controls/ros2_control/issues/2609

christophfroehlich, 2025-10-07:
https://github.com/ros-controls/ros2_control_ci/issues/498

christophfroehlich, 2025-10-23:
https://github.com/ros-controls/ros2_control/issues/2735

christophfroehlich, 2025-10-25:
https://github.com/ros-controls/ros2_control_ci/issues/524

christophfroehlich, 2025-11-04:
https://github.com/ros-controls/ros2_control_ci/issues/533

christophfroehlich, 2025-11-06:
https://github.com/ros-controls/ros2_control/issues/2780
https://github.com/ros-controls/ros2_control_ci/issues/536

christophfroehlich, 2025-11-12:
https://github.com/ros-controls/ros2_control/issues/2806

christophfroehlich, 2025-11-16:
https://github.com/ros-controls/ros2_control_ci/issues/545

christophfroehlich, 2025-12-01:
https://github.com/ros-controls/ros2_control/issues/2871 (rolling/testing) 👀 @saikishor

PR #2557 (2025-09-22): Fix the skipped cycles of controllers running at different rate
Similar to the fix made in https://github.com/ros-controls/ros2_control/pull/2091. I had a similar issue when running controllers at a different rate. 

The following are the test results on our TALOS robot

**With the current state of ros2_control:**
[image]

**With the proposed fix:**
[image]

Should also idealy fix: #1957
PR #2573 (2025-09-22): Fix the skipped cycles of controllers running at different rate (backport #2557)
Similar to the fix made in https://github.com/ros-controls/ros2_control/pull/2091. I had a similar issue when running controllers at a different rate. 

The following are the test results on our TALOS robot

**With the current state of ros2_control:**
[image]

**With the proposed fix:**
[image]

Should also idealy fix: #1957 

<hr>This is an automatic backport of pull request #2557 done by [Mergify](https://mergify.com).