ros-controls/ros2_control #1959: Wrong place to insert the lock

yashi, 2024-12-19:
The PR #1599 which backports #1451 failed to insert the lock in the right place.

https://github.com/ros-controls/ros2_control/blob/6a8053ee1297f02f05758e7c52e76503bda608d3/hardware_interface/src/resource_manager.cpp#L687-L691

This means that when `load_and_initialize_components` is `true` the lock is taken, but the rest of the block will be executed regardless.

@destogl Please confirm.

saikishor, 2024-12-19:
@yashi Can you explain the issue that you are having exactly? That would give us some insight

saikishor, 2024-12-19:
> This means that when `load_and_initialize_components` is `true` the lock is taken, but the rest of the block will be executed regardless.
> Or, is it intended fix?

You are totally right!

yashi, 2024-12-19:
@saikishor Just curious, what exactly are you protecting with `resources_lock_`? I mean, what does _resource_ in the name of the lock refering to? At first I thought `read_write_status.failed_hardware_names` but it's not only this but others as well?

saikishor, 2024-12-19:
Hello!

For Humble, we are trying to protect the read and write methods to access the object before it is constructed completely or configured. 

https://github.com/ros-controls/ros2_control/blob/6a8053ee1297f02f05758e7c52e76503bda608d3/hardware_interface/src/resource_manager.cpp#L1234-L1280

That is the reason, we would need this mutex here. For rolling on, it might not be needed as it is handled individually in each component

christophfroehlich, 2024-12-20:
fixed with #1960

saikishor, 2024-12-20:
The fix in https://github.com/ros-controls/ros2_control/pull/1960 is merged. I will go ahead and close this