ros-controls/ros2_control #1965: Fix deprecation from realtime_tools on humble

christophfroehlich, 2024-12-21:
## Background

We changed the API of lock_memory a bit

```
/workspaces/ros2_humble_ws/src/ros2_control/controller_manager/src/ros2_control_node.cpp: In function ‘int main(int, char**)’:
/workspaces/ros2_humble_ws/src/ros2_control/controller_manager/src/ros2_control_node.cpp:62:50: warning: ‘bool realtime_tools::lock_memory(std::string&)’ is deprecated: Use std::pair<bool, std::string> lock_memory() instead. [-Wdeprecated-declarations]
   62 |   if (lock_memory && !realtime_tools::lock_memory(message))
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
In file included from /workspaces/ros2_humble_ws/src/ros2_control/controller_manager/src/ros2_control_node.cpp:24:
/workspaces/ros2_humble_ws/src/realtime_tools/include/realtime_tools/realtime_helpers.hpp:70:6: note: declared here
   70 | bool lock_memory(std::string & message);
      |      ^~~~~~~~~~~
```

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

kumar-sanjeeev, 2024-12-21:
@christophfroehlich  hi would like to work on this.

christophfroehlich, 2024-12-21:
alright. although this not your first issue anymore :smile:

kumar-sanjeeev, 2024-12-21:
> alright. although this not your first issue anymore 😄

Ahh now I read Instructions carefully, next time I will focus on other bugs or good second issue. Thanks

imsudiproy, 2024-12-22:
Hi @christophfroehlich , I would like to work on this if this is available.

christophfroehlich, 2024-12-22:
@kumar-sanjeeev ok for you taking another one?

kumar-sanjeeev, 2024-12-22:
Yeah sure 👍🏻

imsudiproy, 2024-12-23:
@christophfroehlich @kumar-sanjeeev Thanks

imsudiproy, 2024-12-23:
@christophfroehlich @saikishor [PR](https://github.com/ros-controls/ros2_control/pull/1970) is merged. This issue can be closed now. 
Thanks

saikishor, 2024-12-23:
Closing in favor of #1970 and #1971

closed 2024-12-23 by saikishor