ros-controls/ros2_control #1977: Use urdf/model.hpp instead of deprecated urdf/model.h

christophfroehlich, 2025-01-03:
## Background
We use deprecated header files in the component_parser

```
/opt/ros/rolling/include/urdf/urdf/model.h:35:2: warning: #warning This header is obsolete, please include urdf/model.hpp instead [-Wcpp]
   35 | #warning \
      |  ^~~~~~~
```

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

TimmyMcPickles, 2025-01-04:
I'd like to try working on this issue

nakul-py, 2025-01-04:
Hi @christophfroehlich 
I came across this issue and would like to work on resolving it. Could you please assign it to me?

christophfroehlich, 2025-01-04:
Fyi: This change came in only on rolling, see https://github.com/ros2/urdf/pull/39, i.e., it is missing on jazzy on humble. To come across this, we need to make an include condition. The URDF package does not export a version file, but we can use something like this

```c++
// \note The versions conditioning is added here to support the source-compatibility with Humble and Jazzy
#include "rclcpp/version.h"
#if RCLCPP_VERSION_GTE(29, 0, 0)
#include "urdf/model.hpp"
#elif
#include "urdf/model.h"
#endif
```

PR #1978 (2025-01-07): Using urdf/model.hpp for rolling
### **Description**

**This pull request addresses the use of the deprecated `urdf/model.h` header in component_parser.cpp to ensure compatibility with Rolling, Jazzy, and Humble distributions of ROS 2.**

**Since `urdf/model.h` was replaced by `urdf/model.hpp` in newer ROS 2 distributions starting from Rolling, this PR introduces a version-conditional include to maintain backward compatibility with older distributions.**

Fixes #1977
closed 2025-01-07 by christophfroehlich