ros-controls/ros2_control #1986: std::bad_alloc crash in robot_description_callback

tonynajjar, 2025-01-07:
**Describe the bug**

@saikishor I localized that after this [commit](https://github.com/ros-controls/ros2_control/pull/1567) I get the following crash.
I don't really see the relation though, do you?

**Screenshots**

```
[gazebo-1] [INFO] [1736265253.392840323] [gz_ros_control]: Loading controller_manager
[gazebo-1] [INFO] [1736265253.404793628] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[gazebo-1] [WARN] [1736265253.404938458] [gz_ros_control]: Waiting RM to load and initialize hardware...
[gazebo-1] [INFO] [1736265253.406613475] [controller_manager]: Received robot description from topic.
[gazebo-1] terminate called after throwing an instance of 'std::bad_alloc'
[gazebo-1]   what():  std::bad_alloc
[gazebo-1] Stack trace (most recent call last) in thread 584229:
[gazebo-1] #21   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7ebd568a0c3b, in 
[gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7ebd56813a93, in 
[gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ebd5186edb3, in 
[gazebo-1] #17   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7ebd1a0c0dc6, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[gazebo-1] #16   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7ebd1a0afc29, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[gazebo-1] #15   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7ebd1a0af4fa, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[gazebo-1] #14   Object "/home/angsa/underlay_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7ebd1a4193ef, in rclcpp::Subscription<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void>, std_msgs::msg::String_<std::allocator<void> >, std_msgs::msg::String_<std::allocator<void> >, rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> > >::handle_message(std::shared_ptr<void>&, rclcpp::MessageInfo const&)
[gazebo-1] #13   Object "/home/angsa/underlay_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7ebd1a35516d, in std::__detail::__variant::__gen_vtable_impl<std::__detail::__variant::_Multi_array<std::__detail::__variant::__deduce_visit_result<void> (*)(rclcpp::AnySubscriptionCallback<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::dispatch<std_msgs::msg::String_<std::allocator<void> > >(std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}&&, std::variant<std::function<void (std_msgs::msg::String_<std::allocator<void> > const&)>, std::function<void (std_msgs::msg::String_<std::allocator<void> > const&, rclcpp::MessageInfo const&)>, std::function<void (rclcpp::SerializedMessage const&)>, std::function<void (rclcpp::SerializedMessage const&, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >)>, std::function<void (std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const>)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const> const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage>, rclcpp::MessageInfo const&)> >&)>, std::integer_sequence<unsigned long, 0ul> >::__visit_invoke(rclcpp::AnySubscriptionCallback<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::dispatch<std_msgs::msg::String_<std::allocator<void> > >(std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}&&, std::variant<std::function<void (std_msgs::msg::String_<std::allocator<void> > const&)>, std::function<void (std_msgs::msg::String_<std::allocator<void> > const&, rclcpp::MessageInfo const&)>, std::function<void (rclcpp::SerializedMessage const&)>, std::function<void (rclcpp::SerializedMessage const&, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >)>, std::function<void (std::unique_ptr<std_msgs::msg::String_<std::allocator<void> >, std::default_delete<std_msgs::msg::String_<std::allocator<void> > > >, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const>)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const> const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >)>, std::function<void (std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage>)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage>, rclcpp::MessageInfo const&)> >&)
[gazebo-1] #12   Object "/home/angsa/underlay_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7ebd1a34f993, in std::_Function_handler<void (std_msgs::msg::String_<std::allocator<void> > const&), std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>))(std_msgs::msg::String_<std::allocator<void> > const&)> >::_M_invoke(std::_Any_data const&, std_msgs::msg::String_<std::allocator<void> > const&)
[gazebo-1] #11   Object "/home/angsa/underlay_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7ebd1a34d092, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[gazebo-1] #10   Object "/home/angsa/underlay_ws/install/controller_manager/lib/libcontroller_manager.so", at 0x7ebd1a34c5a6, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[gazebo-1] #9    Object "/opt/ros/jazzy/lib/libgz_ros2_control-system.so", at 0x7ebd3ccd42a6, in 
[gazebo-1] #8    Object "/opt/ros/jazzy/lib/libgz_ros2_control-system.so", at 0x7ebd3ccf8aca, in 
[gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ebd51827ac7, in 
[... cut ...]
[gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7ebd56815b1c, in pthread_kill
[gazebo-1] Aborted (Signal sent by tkill() 583990 1000)
[INFO] [gazebo-1]: process has finished cleanly [pid 583899]

```

**Environment (please complete the following information):**
 - OS: Ubuntu Noble
 - Version: Jazzy
 - ros2_control built from source

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.

saikishor, 2025-01-07:
Hello @tonynajjar,

Thank you for reporting it. Can you please share the ros2_control tag information inside your URDF?. I'm not sure if this is coming from the recent commit, as we have tests internally that are working. I would suggest to remove the installed `hardware_interface` package to just discard some binary incompatibility.

Thank you

saikishor, 2025-01-07:
The above one should work AFAIK. Can you check source only without any binaries of hardware_interface?

> [gazebo-1] #9    Object "/opt/ros/jazzy/lib/libgz_ros2_control-system.so", at 0x7ebd3ccd42a6, in 

Looking at your log, I see that the gz_ros2_control is being used from the binaries. Remember that it is linking againsyt installed binary instead of your local one. Try also compiling this locally

tonynajjar, 2025-01-07:
When building `gz_ros2_control` from source I get:

```
📦 angsa@tony-XMG-CORE-M24:~/underlay_ws/src$ colcon build --packages-up-to gz_ros2_control
Starting >>> launch_ros
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Finished <<< ros2_control_test_assets [0.33s]                                                                                          
Finished <<< controller_manager_msgs [0.79s]                                                      
Finished <<< launch_ros [1.36s]           
Starting >>> joint_limits
Starting >>> launch_testing_ros
Finished <<< joint_limits [0.33s]                                                              
Starting >>> hardware_interface
Finished <<< hardware_interface [0.31s]                                                             
Starting >>> controller_interface
Starting >>> hardware_interface_testing                                                     
Finished <<< hardware_interface_testing [0.25s]                                                                                                     
Finished <<< controller_interface [0.30s]
Finished <<< launch_testing_ros [1.30s]           
Starting >>> controller_manager
--- stderr: controller_manager                               
/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< controller_manager [0.75s]
Starting >>> gz_ros2_control
--- stderr: gz_ros2_control                               
In file included from /opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/serialized.pb.h:12,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/EntityComponentManager.hh:20,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/System.hh:24,
                 from /home/angsa/underlay_ws/src/gz_ros2_control/gz_ros2_control/include/gz_ros2_control/gz_system_interface.hpp:24,
                 from /home/angsa/underlay_ws/src/gz_ros2_control/gz_ros2_control/include/gz_ros2_control/gz_system.hpp:24,
                 from /home/angsa/underlay_ws/src/gz_ros2_control/gz_ros2_control/src/gz_system.cpp:15:
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/serialized.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/serialized.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers. Please
      |  ^~~~~
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/serialized.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/double_v.pb.h:12,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/components/Serialization.hh:31,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/components/Name.hh:23,
                 from /home/angsa/underlay_ws/src/gz_ros2_control/gz_ros2_control/src/gz_ros2_control_plugin.cpp:28:
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/double_v.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/double_v.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers. Please
      |  ^~~~~
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/double_v.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/actor.pb.h:12,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/Conversions.hh:20,
                 from /opt/ros/jazzy/opt/gz_sim_vendor/include/gz/sim8/gz/sim/components/Serialization.hh:37:
/opt/ros/jazzy/opt/gz_msgs_vendor/include/gz/msgs10/gz/msgs/details/actor.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
...
```

I guess I would also need to build gz_msgs_vendor/gz_msgs from source which doesn't seem so simple (or at least I don't know how because it's vendored). Do you know a way?

This is the installed version:

```
ros-jazzy-gz-msgs-vendor is already the newest version (0.0.5-1noble.20241218.083446).
```

saikishor, 2025-01-07:
@tonynajjar are you using the [jazzy](https://github.com/ros-controls/gz_ros2_control/tree/jazzy) branch?. If so, try with the last tag and see if it works

tonynajjar, 2025-01-07:
sorry forgot to mention, yes I'm using the jazzy branch. I don't want to take too much of your time on what I believe is like you say, an issue with my setup e.g binary mismatch. I will try a bit more and if I can't get it to work I'll just pin my fork to before that problematic commit.

saikishor, 2025-01-07:
Sure. No worries. Thank you. Keep us posted :)

saikishor, 2025-04-02:
@tonynajjar do you still have this issue?

closed 2025-04-02 by tonynajjar

SKrishna1340, 2025-05-21:
I recently started getting this error from a project that was working earlier. So, I tried a couple of different methods to isolate the problem.
- I tried isolating the nodes I am creating in my launch file, and it seems the spawner script `(ros2 run ros_gz_sim create ...)` is causing this error.
- It is not a problem with the URDF or the launch file. I tried running the pre-built gazebo tutorial example (cart position control), and got the same error messages.

 @tonynajjar: If you remember any posts/steps that helped you resolve this issue, please share them. Thank you.

My error log for reference:
[INFO] [launch]: All log files can be found below /home/mskrishna/.ros/log/2025-05-20-22-27-57-143615-saikm-dellg15-45808
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [45812]
[INFO] [parameter_bridge-2]: process started with pid [45813]
[INFO] [robot_state_publisher-3]: process started with pid [45814]
[INFO] [create-4]: process started with pid [45815]
[robot_state_publisher-3] [INFO] [1747805277.676387883] [robot_state_publisher]: Robot initialized
[create-4] [INFO] [1747805277.677575689] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-2] [INFO] [1747805277.687171283] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[create-4] [INFO] [1747805278.228006562] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-4] [INFO] [1747805278.327488536] [ros_gz_sim]: Entity creation successful.
[INFO] [create-4]: process has finished cleanly [pid 45815]
[INFO] [spawner-5]: process started with pid [45945]
[spawner-5] [INFO] [1747805278.833127645] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[gazebo-1] [INFO] [1747805279.894248209] [gz_ros_control]: [gz_ros2_control] Fixed joint [world_to_base] (Entity=17)] is skipped
[gazebo-1] [INFO] [1747805279.897901307] [gz_ros_control]: Loading controller_manager
[gazebo-1] [INFO] [1747805279.912684033] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[gazebo-1] [WARN] [1747805279.914077909] [gz_ros_control]: Waiting RM to load and initialize hardware...
[gazebo-1] [INFO] [1747805279.914211289] [controller_manager]: Received robot description from topic.
[gazebo-1] terminate called after throwing an instance of 'std::bad_alloc'
[gazebo-1]   what():  std::bad_alloc
[gazebo-1] Stack trace (most recent call last) in thread 45984:
[gazebo-1] #19   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x764b7ab29c3b, in 
[gazebo-1] #17   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x764b7aa9caa3, in 
[gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758ecdb3, in 
[gazebo-1] #15   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x764b41f1e206, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[gazebo-1] #14   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x764b41f0c639, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[gazebo-1] #13   Object "/opt/ros/rolling/lib/librclcpp.so", at 0x764b41f0befa, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[gazebo-1] #12   Object "/opt/ros/rolling/lib/libcontroller_manager.so", at 0x764b4225b084, in 
[gazebo-1] #11   Object "/opt/ros/rolling/lib/libcontroller_manager.so", at 0x764b421e823a, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[gazebo-1] #10   Object "/opt/ros/rolling/lib/libcontroller_manager.so", at 0x764b421e79fe, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[gazebo-1] #9    Object "/home/mskrishna/moveit2_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so", at 0x764b4230ca7c, in gz_ros2_control::GZResourceManager::load_and_initialize_components(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)
[gazebo-1] #8    Object "/home/mskrishna/moveit2_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so", at 0x764b422e4fd9, in 
[gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758a5ac7, in 
[gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758bb390, in __cxa_throw
[gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758a5a54, in std::terminate()
[gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758bb0d9, in 
[gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x764b758a5ff4, in 
[gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x764b7aa288fe, in abort
[gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x764b7aa4527d, in gsignal
[gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x764b7aa9eb2c, in pthread_kill
[gazebo-1] Aborted (Signal sent by tkill() 45863 1000)

christophfroehlich, 2025-05-21:
>     * It is not a problem with the URDF or the launch file. I tried running the pre-built gazebo tutorial example (cart position control), and got the same error messages.

which versions of ros2_control and gz_ros2_control are you running on which distro? E.g., share the output of `ros2 pkg xml -t version gz_ros2_control`

SKrishna1340, 2025-05-22:
Hi @christophfroehlich, here are my system details:
- I am using Ubuntu 24.04 with ROS2 Rolling, and GZ Ionic.
- Versions for the packages you requested.
 - ros2_control: 4.27.0
 - gz_ros2_control: 2.0.1