ros-controls/ros2_control #2008: Remapping arguments deprecation warning also shown when correctly used

robin-zealrobotics, 2025-01-20:
The new `--controller-ros-args`-related deprecation warning is shown also when passing the remaps through the spawners as the deprecation warning suggests.

saikishor, 2025-01-20:
Hello!

Thank you for reporting the issue. Do you mind sharing the log?

Thank you

saikishor, 2025-01-20:
Hello!

Ideally, the deprecation notice should only be shown when the remapping is done through the `ros2_control_node` 
https://github.com/ros-controls/ros2_control/blob/54a9226655eb330759cda21d0445e199cb25a53a/controller_manager/src/controller_manager.cpp#L3508-L3518

If you can share more logs, it would be great,.

Thank you!

saikishor, 2025-02-03:
> Given that these are not remaps intended for the spawner-nodes specifically, but generally for all nodes, probably we wouldn't want deprecation warnings being printed out.
> Not sure if it's easy to determine though which kinds of remaps should print the deprecation and which should not.

We do understand that it is hard to differentiate which type of remaps belongs to what, that's why the depreciation is printing irrespective of what it belongs to. We want users to use the new ways to remap for controllers and we don't want it to be a breaking change. 

The issue you are mentioning should be solved soon in some upcoming releases for rolling