ros-controls/ros2_control #2010: value_ptr_ issue when copying/moving hardware_interface::Handle

kimsh5904, 2025-01-21:
**Describe the bug**
- It seems `value_ptr_` points the same `value_` after calling the below operators

https://github.com/ros-controls/ros2_control/blob/d4139b7a4943844b7e61ef59a3d81d8d8beb4baa/hardware_interface/include/hardware_interface/handle.hpp#L75-L125

saikishor, 2025-01-21:
Hello @kimsh5904 !

Thank you for reporting the issue. Let me check and get back to you. If you already have a minimal example, it already helps to reproduce it

Wiktor-99, 2025-01-21:
I think what we can see here is wrong ptr copy, I'm not sure but maybe we wold like to to something like this:
```cpp
Handle(const Handle & other) noexcept 
 { 
   std::unique_lock<std::shared_mutex> lock(other.handle_mutex_); 
   std::unique_lock<std::shared_mutex> lock_this(handle_mutex_); 
   prefix_name_ = other.prefix_name_; 
   interface_name_ = other.interface_name_; 
   handle_name_ = other.handle_name_; 
   value_ = other.value_; 
   value_ptr_ = &value_; 
 } 
``` 
We would like copy only the value but the pointer should still point to our member value_ not to member of the other class

saikishor, 2025-01-21:
I've added some tests here: https://github.com/ros-controls/ros2_control/pull/2011

The changes look good to me. See if you can find some issues in the tests. I'm happy to check back and fix them. 

Thank you for your time guys! :)

closed 2025-02-07 by bmagyar