ros-controls/ros2_control #2012: Segfault on Ctrl-C

mjforan, 2025-01-22:
**Describe the bug**
Killing a `ros2_control` launch with Ctl-C causes a segfault and cleanup methods like `on_deactivate` do not run.

**To Reproduce**
```
git clone https://github.com/ros-controls/ros2_control_demos.git
cd ros2_control_demos
docker build . -f Dockerfile/Dockerfile -t ros2_control_demos --build-arg ROS_DISTRO=jazzy
docker run --rm -it ros2_control_demos bash
. install/setup.bash
ros2 launch ros2_control_demo_example_1 rrbot.launch.py
```
wait for everything to initialize...
Ctrl-C

**Expected behavior**
Clean shutdown

**Screenshots**
```
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-2] [INFO] [1737557728.284547701] [rclcpp]: signal_handler(signum=2)
[ros2_control_node-1] [INFO] [1737557728.284551321] [rclcpp]: signal_handler(signum=2)
[ros2_control_node-1] Stack trace (most recent call last):
[ros2_control_node-1] #7    Object "/usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2", at 0xffffffffffffffff, in 
[ros2_control_node-1] #6    Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x64e5c5265c94, in _start
[ros2_control_node-1] #5    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x71a0f848f28a, in __libc_start_main
[ros2_control_node-1] #4    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x71a0f848f1c9, in 
[ros2_control_node-1] #3    Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x64e5c5264ce2, in main
[ros2_control_node-1] #2    Object "/opt/ros/jazzy/lib/controller_manager/ros2_control_node", at 0x64e5c5265eb2, in 
[ros2_control_node-1] #1    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x71a0f8a596b9, in rclcpp::Executor::~Executor()
[ros2_control_node-1] #0    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x71a0f8a52e51, in 
[ros2_control_node-1] Segmentation fault (Address not mapped to object [0x71a0f45fd020])

```

**Environment (please complete the following information):**
 - OS: [Ubuntu 24.04]
 - Version [Jazzy]

christophfroehlich, 2025-01-22:
Thanks for reporting this. I thought that I had reported that already, but can't find any similar issues now.

I saw that in the tests, that's why I have deactivated the return code check
https://github.com/ros-controls/ros2_control_demos/blob/master/example_1/test/test_rrbot_launch.py#L149

christophfroehlich, 2025-01-22:
https://github.com/ros-controls/ros2_control_demos/pull/548

mjforan, 2025-01-22:
To be clear, this is not only an issue with the demos. I see this new behavior on my own hardware interfaces as well.

christophfroehlich, 2025-01-22:
Sure, but I saw that with the tests/demos already -> we can reproduce it and work on that.

christophfroehlich, 2025-01-22:
@saikishor Interestingly, we don't see this in the test [here in this repo](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_ros2_control_node_launch.py#L134).

saikishor, 2025-01-22:
> @saikishor Interestingly, we don't see this in the test [here in this repo](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/test/test_ros2_control_node_launch.py#L134).

That's weird. If it can be easily reproducable, then should be easy to fix

mjforan, 2025-01-22:
I was working in a slightly older container and got a different result on ctrl-c. Maybe this will help guide your debugging:
```
[ros2_control_node-2] /opt/ros/jazzy/lib/controller_manager/ros2_control_node: symbol lookup error: /opt/ros/jazzy/lib/libdiff_drive_controller.so: undefined symbol: _ZN20controller_interface23ControllerInterfaceBaseD2Ev
```

And here are some relevant software versions:
```
ros-jazzy-controller-interface/now 4.21.0-1noble.20241228.025831 arm64 [installed,local]
ros-jazzy-controller-manager/now 4.21.0-1noble.20241228.041458 arm64 [installed,local]
ros-jazzy-diff-drive-controller/now 4.19.0-1noble.20250115.225929 arm64 [installed,local]
```

saikishor, 2025-01-22:
Nothing has been done to fix this in the last releases. I doubt that this is fixed. 

Thanks. I'll check it out.

saikishor, 2025-01-24:
Hello @mjforan!

This PR : https://github.com/ros-controls/ros2_control/pull/2014 should fix the issue you are facing. It would be great, if you can try it out and If it works, please approve the PR.

Thank you!