ros-controls/ros2_control #2020: Invalid size of memlock

KmakD, 2025-01-29:
**Describe the bug**
According to documentation the memlock should be set to 102400 bytes:

```
@realtime soft memlock 102400
@realtime hard memlock 102400
```

With such settings however, when starting controller manager following warning appears:
```
[controller_manager]: Unable to lock the memory: 'No proper privileges to lock the memory!'
```

If I change the limits to "unlimited", then the warning doesn't appear. Looks like the controller_manager is trying to lock memory with a greater value than 102400 bytes (higher than "hard" memlock value).

I've tested it also in docker with "docker way" of setting ulimits and the result is the same. Warning appears for value 102400 and not when "unlimited".

**To Reproduce**
Steps to reproduce the behavior:
1. Adjust limits in  `/etc/security/limits.conf` 
2. Run controller manager
3. Warning appears

**Expected behavior**
No warning, memory locked properly.

**Environment:**
 - OS: Ubuntu
 - Version: Humble (latest apt release)
 - RPi 4 with custom real-time kernel

mrjogo, 2025-02-14:
I had the same issue (launching via `ur_robot_driver`) on Ubuntu 24.04 Jazzy w/ PREEMPT_RT. I ended up just increasing the limit by 10x (to `1024000`)

mrjogo, 2025-02-14:
Juuuuuust kidding. With the larger value the node starts correctly, but then quickly crashes with `std::bad_alloc`. `unlimited` it is.

```bash
[ros2_control_node-2] [INFO] [1739568863.511034167] [controller_manager]: Received robot description from topic.
[ros2_control_node-2] terminate called after throwing an instance of 'St9bad_alloc'
[ros2_control_node-2]   what():  std::bad_alloc
[ros2_control_node-2] terminate called recursively
```

saikishor, 2025-02-14:
With `unlimited` it should work as expected. If you are still having issues, you can set the `lock_memory` parameter to false.

I can confirm that for us with `unlimited` it works

KmakD, 2025-02-17:
Should the docs for CM be updated before we close this issue?

saikishor, 2025-02-17:
> Should the docs for CM be updated before we close this issue?

Hello @KmakD 

It is already documented here : https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#launching-controller-manager-with-ros2-control-node

Can you check it?

Thank you

KmakD, 2025-02-17:
This section describes how to set limits and it still has old values for memlock: https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/userdoc.html#determinism

christophfroehlich, 2025-02-20:
@KmakD could you open a PR in fixing this please?

closed 2025-02-24 by christophfroehlich