ros-controls/ros2_control #2031: Update docs about semantic_components

christophfroehlich, 2025-02-03:
We have an (outdated) list of semantic components here: 
https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/hardware_interface_types_userdoc.html#sensors
Where should we put this in the docs (can be a follow-up PR)?

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_control/pull/1945#pullrequestreview-2582429564_

christophfroehlich, 2025-02-03:
@tpoignonec what do you think?

tpoignonec, 2025-02-03:
What about adding a subsection in guidelines and best practices?

**Example text:**

> #Hardware Components
> ##Guidelines and Best Practices
> ###Using semantic components
>
> In order to streamline the configuration of commonly used hardware interface, so-called _semantic components_ can be used to wrap mechanisms to claim / release the interfaces. The base components `semantic_components::SemanticComponentInterface` and `semantic_components::SemanticComponentCommandInterface` are used to define semantic components for read-only and write only devices, respectively.
>
> List of existing  `SemanticComponentInterface` and associated broadcasters, if any:
> -  `IMUSensor`, used by  [Imu Sensor Broadcaster](https://control.ros.org/humble/doc/ros2_controllers/imu_sensor_broadcaster/doc/userdoc.html#imu-sensor-broadcaster-userdoc)
> - `ForceTorqueSensor`, used by [Force Torque Sensor Broadcaster](https://control.ros.org/humble/doc/ros2_controllers/force_torque_sensor_broadcaster/doc/userdoc.html#force-torque-sensor-broadcaster-userdoc)
> - `GPSSensor`
> - `PoseSensor`
> - `RangeSensor`

> List of existing  `SemanticComponentCommandInterface` and associated broadcasters, if any:
> - `LedRgbDevice`

**Location in docs:**

[image]

PR #2032 (2025-02-04): Semantic components docs
Implements and fixes #2031 
[image]
closed 2025-02-04 by christophfroehlich