ros-controls/ros2_control #2039: Trouble with the controller_manger since this year

LukasLange28, 2025-02-06:
**Describe the bug**
The controller_manager hangs up at "Loading controller '<spawned_controller>'"
The spawner does not get a result from calling "/controller_manager/load_controller"

**To Reproduce**
1. Run the ros2_control_node with a yaml file (In the yaml file is a joint_state_broadcaster and a ackermann_steering_controller)
2. Run a robot_state_publisher with a robot_description
3. Run the spawner from controller_manager to spawn the joint_state_broadcaster an the ackermann_steering_controller
4. Here does the controller_manager hang up

**Expected behavior**
The two controllers should spawn

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version Humble
 - ros2_control: 2.47.0
 - ros2_controllers: 2.41.0

**Additional context**
The robot was running successfully in up to December 2024.
Since then the robot was untouched.
Now (In February) I want to continue on the robot and had trouble to start the system that already worked before.
Because the system was not touched in the time between, I checked the versions of the ros2_control and ros2_controller packages.
Both are on the newest release.
If I checked it correct, both packages had releases on Humble.
Because the controller_manager hangs up, I expect the problem there.
Maybe some changes from ros2_control 2.45.0 to 2.47.0 causes the trouble.

LukasLange28, 2025-02-06:
**Update:** 

1. If I spawn a controller, the node of the controller is getting started and runs.
The controller manager can still be found with "ros2 node list". So it hangs up after spawning the controller but does not crash.
2. The behaviour is the same if I use the load_controller service direct.

christophfroehlich, 2025-02-10:
Can you confirm that the [demos](https://control.ros.org/humble/doc/ros2_control_demos/doc/index.html#examples) work? 

Can you please add the full log output of your system?

LukasLange28, 2025-02-11:
I was able to solve it.
In my ros2_controllers.yaml was a "use_sim_time: true" in the controller_manager ros__parameters.
That didn't cause any trouble in the past.
If I delete this line, everything works fine.

closed 2025-02-11 by LukasLange28

christophfroehlich, 2025-02-11:
we fixed something in the ros2_control_node to handle sim_time properly. it may be that it hadn't had an effect before, but now it does. glad you found it