ros-controls/ros2_control #2041: humble: An example for parameters_file to be processed by determine_controller_node_options

tenfoldpaper, 2025-02-07:
I'm currently developing a mujoco-ros2_control integration project in humble. 

I've been dealing with the issues of global node name remappings, as the ros2_control is designed as a plugin that is loaded to the main mujoco_ros node.

The remapping tutorial from the docs (https://control.ros.org/humble/doc/ros2_control/controller_manager/doc/userdoc.html) has been helpful for changing the controller manager's node name, but now I'm trying to address the issue of remapping the individual controller's nodes. 

I saw that there has been an active discussion dealing with this in the `master` branch of `ros2_control_demos`, but it seems that those changes haven't been backported into `humble`.

Still, I noticed that there is the function `determine_controller_node_options`, which is available for the humble branch. There, it is possible to modify the node options via `controller.info.parameters_files` -- but I haven't been able to find any examples on how this function is supposed to work, such as how to define the parameter files for specific controllers.

Would it be possible to add some instructions on how this can be done, using a common controller like the `joint_state_broadcaster`? I'd be happy to add a cleaned up documentation if the solution is provided as an answer to the issue.

bmagyar, 2025-02-07:
@saikishor is this functionality fully fleshed out in Humble or the API we have there is more of the beginnings of this functionality?

saikishor, 2025-02-07:
Hello @tenfoldpaper !

For Humble, with the controller.info.parameter_files you can only pass a set of parameter files to be loaded by the controllers. However, I'm afraid you cannot change the node naming or other attributes at runtime. 

Recently, on rolling we added a capability to spawners with option `--controller-ros-args` where you can parse node options specific to each controller like topic or service remapping and more. I'm afraid this is not backportable. 

If you are interested to check this feature of rolling and Jazzy, you can compile the rolling branch on Humble using the instructions presented here : https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source.

I hope this answers your question.