ros-controls/ros2_control #2049: After upgrading ros2_control, controller_manager is unable to load controllers

Fufs, 2025-02-13:
**Describe the bug**
The controller_manager attempts to load a controller but just stops and doesn't seem to do anything. The controller_manager from gz_ros2_control (in Gazebo) still works fine and loads the controllers.

**To Reproduce**
Launch file
```py
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.conditions import IfCondition
from launch.substitutions import LaunchConfiguration
from launch.actions import DeclareLaunchArgument, RegisterEventHandler
from launch.event_handlers import OnProcessExit

def generate_launch_description():

    pkg_share = get_package_share_directory("luggage_av")

    start_controller = LaunchConfiguration("start_controller")
    namespace = LaunchConfiguration("namespace")

    controller_manager = Node(
        package="controller_manager",
        executable="ros2_control_node",
        # arguments=["--ros-args","--log-level","debug"],
        parameters=[
            os.path.join(pkg_share, "parameters", "controller_manager.yaml"),
        ],
        output="screen",
        namespace=["/", namespace],
        condition=IfCondition(start_controller)
    )

    joint_state_broadcaster_spawner = Node(
        package='controller_manager',
        executable='spawner',
        arguments=[
            "joint_state_broadcaster",
            "--controller-ros-args", "--remap /tf:=tf"
        ],
        namespace=["/", namespace]
    )

    diff_drive_controller_spawner = Node(
        package='controller_manager',
        executable='spawner',
        arguments=[
            'diff_drive_controller',
            '--param-file', os.path.join(pkg_share, "parameters", "diff_drive_controller.yaml"),
            "--controller-ros-args", "--remap /tf:=tf --remap diff_drive_controller/cmd_vel:=cmd_vel",
        ],
        namespace=["/", namespace]
    )

    return LaunchDescription([
        DeclareLaunchArgument(
            "start_controller",
            default_value="true"
        ),
        DeclareLaunchArgument(
            "namespace",
            default_value="luggage_av",
            description="Namespace of the bot (usually its unique identifier)"
        ),

        controller_manager,
        joint_state_broadcaster_spawner,
        RegisterEventHandler(
[... 68 lines cut ...]
[spawner-5]     sys.exit(load_entry_point('controller-manager==4.25.0', 'console_scripts', 'spawner')())
[spawner-5]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[spawner-5]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 200, in main
[spawner-5]     if is_controller_loaded(
[spawner-5]        ^^^^^^^^^^^^^^^^^^^^^
[spawner-5]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/spawner.py", line 67, in is_controller_loaded
[spawner-5]     controllers = list_controllers(
[spawner-5]                   ^^^^^^^^^^^^^^^^^
[spawner-5]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/controller_manager_services.py", line 172, in list_controllers
[spawner-5]     return service_caller(
[spawner-5]            ^^^^^^^^^^^^^^^
[spawner-5]   File "/opt/ros/jazzy/lib/python3.12/site-packages/controller_manager/controller_manager_services.py", line 150, in service_caller
[spawner-5]     raise RuntimeError(
[spawner-5] RuntimeError: Could not successfully call service /controller_manager/list_controllers after 3 attempts.
[ERROR] [spawner-5]: process has died [pid 187560, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner diff_drive_controller --param-file /path/to/ros2_ws/install/luggage_av/share/luggage_av/parameters/diff_drive_controller.yaml --controller-ros-args --remap /tf:=tf --remap diff_drive_controller/cmd_vel:=cmd_vel --ros-args -r __ns:=/'].

```

**Environment (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu 24.04LTS
 - Version [e.g. Foxy]: Jazzy
- Anything that may be unusual about your environment

**Additional context**
Add any other context about the problem here, especially include any modifications to ros2_control that relate to this issue.

christophfroehlich, 2025-02-13:
> /path/to/ros2_ws/
Have you altered the path for this issue, right?

Can you test [example_2](https://control.ros.org/jazzy/doc/ros2_control_demos/example_2/doc/userdoc.html), and if it works compare it with your config?

Fufs, 2025-02-13:
Correct, it contained sensitive info. 

I'll test example_2 and report back in a moment, but before I do, here's some additional context:
* the log was generated by running the launch file with launch argument namespace:="" to generalize the issue and make it searchable when a solution is found. Both the default and custom namespaces had this issue.
* the logs say the update rate is 200Hz, where the parameter is set to 100Hz. This is because I tried changing it to see if the parameters from the YAML are being used (I had issues before where using hardcoded namespaces in the old style namespace: {controller_manager:{...}} would prevent the params from being applied due to a change in the parser logic [here](https://github.com/ros-controls/ros2_control/blob/254b6e85445cc4425d8400518a2cbb607ea0f9c1/controller_manager/controller_manager/controller_manager_services.py#L324-L336)). I had since reverted the update rate but used the logs where I confirmed the parameters file were loading correctly.
* Peaking at service list (`ros2 service list`) the service `/controller_manager/load_controller` does indeed exist. Not sure why calling it fails.
* Using ros2_control provided by apt, made sure to update to fully upgrade all the packages before submitting this issue

Fufs, 2025-02-13:
Ok, I modified the example_2 to use my launch file, and formatted its params files to resemble mine. The manager loads the controllers, however, now I'm running into a different issue, where the spawner for diff_drive_controller is not using the `--param-file` argument to pass another params file when spawning the controller. I will look more into this in the evening. And then compare the 2 packages together.

Fufs, 2025-02-14:
The issue with --param-file not being loaded was a mistake on my part. In my second file, I have copied over the controller name as diffbot_base_controller instead of diff_drive_controller, which I was using in my launch file. The spawner did not include that file as it didn't find the right key in it. It would be nice if there was a warning about trying to explicatively append another params file without a valid key within it, instead of just ignoring it.

All the controllers are started now correctly in the demo, I'll compare the differences and report back.

Fufs, 2025-02-14:
I added hardware mocking to our package and the controllers are loaded correctly with the `GenericSystem`. It seems to be a hardware issue.

Fufs, 2025-02-14:
It was a hardware issue, I was opening the wrong serial port. Sorry for the troubles.

christophfroehlich, 2025-02-14:
glad you found it