ros-controls/ros2_control #2051: Using a RT-safe memory allocator

christophfroehlich, 2025-02-14:
> Question: would using an RT-safe memory allocator not be a more scalable approach?

@gavanderhoorn Do you have some pointers for us? (no pun intended! :D )

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_control/issues/1801#issuecomment-2449726268_

gavanderhoorn, 2025-02-18:
On [Implementing a custom memory allocator](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Allocator-Template-Tutorial.html) there is a (tiny) section about [using TLSF](https://docs.ros.org/en/jazzy/Tutorials/Advanced/Allocator-Template-Tutorial.html#the-tlsf-allocator).

yassine-cherni, 2025-04-08:
Hello, i've made this PR , [https://github.com/ros-controls/ros2_control/pull/2169](url) it need to be improved little bit, we still need some Pointers ;'')