ros-controls/ros2_control #2064: Rework docstrings to have consistent style

christophfroehlich, 2025-02-20:
It would be great to be consistent with the style of the existing docstrings, i.e. start the sentence with a verb for "brief". For example, I would put it here

```cpp
/**
 * @brief Publish the state of the controller manager.
 * 
 * The state includes the list of controllers and the list of hardware interfaces along with their states.
 */
```

However, I realize that we have already different docstring styles in this file. An example of another style is:

```cpp
/// A method to register a callback to be called when the list is switched
/**
* \param[in] callback Callback to be called when the list is switched
*/
```

Is it intentional to have these different styles? @christophfroehlich

_Originally posted by @Maverobot in https://github.com/ros-controls/ros2_control/pull/2006#discussion_r1921631365_

Vishwanath-Karne, 2025-03-02:
Hello @christophfroehlich,

I have worked on this issue and submitted a PR: [#2087][(https://github.com/Vishwanathkarne/pull/2087) ](https://github.com/ros-controls/ros2_control/pull/2087#issue-2889722330).

- I have ensured a consistent docstring style across the codebase.
- Followed the suggested format with @brief starting with a verb.
- Standardized different styles to improve clarity and maintainability.

This is my 1st open-source contribution, and I would love your feedback! 
Let me know if any improvements are needed. 
Thanks for this opportunity!
@christophfroehlich @saikishor 

Regards,
Vishwanath Karne

christophfroehlich, 2025-07-02:
there was no response in the PR since March from @im23123v?

ussefdesouky, 2025-08-19:
Hi @christophfroehlich ,

Is there a space for contribution on this issue?
I am ready to start!